lora-aprs/LoRa_APRS_iGate #3: where is LORA_SCK, LORA_MISO, LORA_MOSI, LORA_CS

zitanjian, 2020-05-30:
void setup_lora()
{
	Serial.println("[INFO] Set SPI pins!");
	SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_CS);
	LoRa.setPins(LORA_CS, LORA_RST, LORA_IRQ);

LORA_SCK, LORA_MISO, LORA_MOSI, LORA_CS);--------------？
	LoRa.setPins(LORA_CS, LORA_RST, LORA_IRQ);-----------？

zitanjian, 2020-05-30:
https://item.taobao.com/item.htm?spm=a1z09.2.0.0.72a92e8dcCgZea&id=614391294183&_u=uv17ec44

zitanjian, 2020-05-30:
https://heltec-automation-docs.readthedocs.io/en/latest/esp32/index.html

peterus, 2020-05-31:
If you are looking for a definition for the constants: they are defined in the Arduino Library for the chip.

Like for the TTGO board: https://github.com/espressif/arduino-esp32/blob/master/variants/ttgo-lora32-v1/pins_arduino.h

The TTGO LoRa32 is pin compatible with the one from heltec.

zitanjian, 2020-07-07:
https://item.taobao.com/item.htm?spm=a1z09.2.0.0.72a92e8dcCgZea&id=614391294183&_u=uv17ec44
//            SX1278    ESP32
// NSS，CS:     19     GPIO 18        片选*
// DIO0:        8      GPIO 26        中断*，支持模拟
// RESET:       7      GPIO 14        复位*，（支持模拟）
// DIO1:        9      GPIO 35       数据输入输出，支持模拟
// DIO2:        10     GPIO 34       数据输入输出，支持模拟，FSK模拟使用
// MOSI：       18     GPIO 27       主出从进，SPI数据输入，（支持模拟）
// MISO：       17     GPIO 19       主进从出，SPI数据输出
// SCK：        16     GPIO 5         SPI时钟
//RXTX：     20                           Rx/Tx开关，Tx高
//BOOST：  27                           大功率输出

closed 2020-07-07 by zitanjian