lora-aprs/LoRa_APRS_iGate #4: Invalid APRS-IS used in LoRa_APRS_iGate.cpp

DL2MF, 2020-05-30:
As far as I can see, there is an issue with commit 8489be29c97b8dbcf4ed9a738e7b1c4039543635, violating the APRS-IS regulations from Bob Bruninga, WB4APR.

In this commit 8489be29c97b8dbcf4ed9a738e7b1c4039543635 src/LoRa_APRS_iGate.cpp has been changed (from also invalid "APRS") to:
`msg.setDestination("APLG0");`

Quote from the APRS-IS code of conduct (http://www.aprs-is.net/):
**Authors, do NOT hijack another software's TO call (APxxxx). If you don't understand APRS and APRS-IS, don't design software to falsely identify itself.**

There is an explizit definition for testing software and other purposes:
 APZ  APZxxx  Experimental

Please change this to keep the APRS-IS network running like it has been designed.

mni tnx!
Mike, DL2MF

peterus, 2020-05-31:
Thanks for the message! As you can read, it is about hijacking other to-calls which are already used.
I checked the official file and the one which is used in the firmware is not in use.

I definitely will register the iGate and Tracker Firmware with Bob.

peterus, 2020-08-13:
Finally had the time to write Bob an Mail and just 2 hours later it is already done: http://aprs.org/aprs11/tocalls.txt
Will change the to-call to a correct format in the next push.

Thanks for pointing this out!

closed 2020-08-13 by peterus