lora-aprs/LoRa_APRS_iGate #54: Mobile digi?

drtechga, 2021-01-03:
Could a mobile digi option be added to pull the gps from the module to use as the digi's location. For instance a digi/tracker located in car.

peterus, 2021-01-03:
sorry i am not understanding your question, can you please describe your idea a little bit better? thanks!

drtechga, 2021-01-04:
[image]
 I attached a diagram to help describe the option requested.

quarksteilchen, 2021-01-04:
I think what @drtechga meant was this:

Add an option to fetch the iGates' location not only from the Config, but also doing a live-update and fetch the location from the built in GPS-module.

```
String lat = create_lat_aprs(Config.beacon.positionLatitude);
String lng = create_long_aprs(Config.beacon.positionLongitude);
```

would then be something like: (shown only for Lat, for Lon it's the same)

```
double lat = Config.beacon.positionLatitude;
if(Config.beacon.doLivePositionUpdate)
    lat = fetchPosFromGPSModule_lat();
String lat = create_lat_aprs(lat);
```

jgarruba, 2021-01-06:
Quark While i have been wanting to do what you just showed in code i don't think that is what drtech means. 

I think he is asking for something the system already does as part of digipeater. He means can the tracker in the car relay data to the igate for the dog and the person. 

As far as I know this already happens is digipete is set to true

peterus, 2021-05-14:
now i am understanding what you mean! the iGate should have a static location and no gps tracking etc. should happen. This ticket is better in the Tracker repository.

closed 2021-05-14 by peterus