lora-aprs/LoRa_APRS_iGate #59: KISS TNC functionality

SQ9MDD, 2021-01-15:
I report the need for KISS-TNC mode for use with APRS applications. 
For example with APRSIS32, APRX or APRSDROID software

I would like to connect DIGI LORA with DIGI APRS in 2m band which works under APRX control, so that data from LORA can be sent directly to APRS bypassing APRS-IS for off-grid operation

M0IGA, 2021-01-15:
Agree with SQ9MDD we need KISS TNC. It will help to build LoRa based full working DIGIS.

SQ2iBK, 2021-01-18:
Dears,
We will be very grateful for your help in writing kissTNC.

Kind Regards,
SQ2IBK

peterus, 2021-03-13:
as there is already a working KISS-TNC implementation from @SQ9MDD  i will close this ticket.
https://github.com/SQ9MDD/TTGO-T-Beam-LoRa-APRS

closed 2021-03-13 by peterus