lora-aprs/LoRa_APRS_iGate #66: WXStation on iGate by BMP/BME280 sensor

faridfarhan28, 2021-02-20:
Hi, could you add some feature Weather Station by BMP/BME280 sensor on iGate like picture on bellow ?
these name is CWOP APRS. Thanks

[image]

faridfarhan28, 2021-02-20:
maybe gived optional sensor
BMP/BME280 (temperature/huminity/Aviation QNH Pressure (mbar))
DHT (temperature/Huminity)

faridfarhan28, 2021-02-20:
CWOP APRS format refrence

http://pond1.gladstonefamily.net/aprswxnet.html

peterus, 2021-02-26:
It will be as an enhancement listed in my head ;) currently there is more important stuff to fix, thanks for your idea!

peterus, 2022-05-07:
as I decided I want to focuse on a "simple" but powerful iGate this is no feature which I would like to add.

closed 2022-05-07 by peterus