lora-aprs/LoRa_APRS_iGate #69: Hi Peter,

DM5TU, 2021-02-25:
I am running the iGate battery powered and buffered by a solar panel. 

If the battery runs empty bcu the solar energy is not enough to reload the battery, the iGate reports the ERROR showed on the picture. 

I was able to reproduce that failure repeatedly. 

Maybe it is a bug in software. 

[image]

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: 
 - PlatformIO Version: 
 - Firmware Version: 

**Board name:**
- 

**Additional context**
Add any other context about the problem here.
73 de sTef DM5TU

DM5TU, 2021-02-25:
@peterus 
I was able to connect the board to the PC tonight.
I found some issues in the VSC.
Maybe its helpful somehow.

I needed to upload the sketch again.

Now its running again.
Let me know if you like to do some testing .

73 de sTef DM5TU
[image]

peterus, 2021-02-26:
thanks for showing me this. unfortunately this will be very tricky to solve and i think this will be an issue with the chip. the settings are stored in the flash which could get corrupted because of the to low voltage.
Is it possible to add some HW protection which is outputing the 5v or nothing?

the issues which are shown in the picture are not really issues with the firmware. VSC is just trying to get some stuff together but doesn't have all information from platformio. if the compile is okay, it will work ;)

peterus, 2021-05-14:
@DM5TU is this issue still existing with the latest firmware?

closed 2022-05-07 by peterus