lora-aprs/LoRa_APRS_iGate #73: Implement Base91 and/or mic-e

peterus, 2021-03-14:
http://www.aprs.org/aprs12/mic-e-examples.txt

BD5HTY, 2021-03-30:
Hello peterus,
1833net@163.com This is my email address, please contact me if possible, thank you!

peterus, 2021-03-30:
> Hello peterus,
> This is my email address, please contact me if possible, thank you!

I have the feeling that you are a bot, so i removed your email. if you have questions please write the questions in github. if you will not response in the next 48h i will remove you post.

BD5HTY, 2021-03-31:
Sorry for any inconveniences caused. I have been looking for your contact information for a long time, but I can't find it. I am from China and I have provided you with a small sponsorship. I would like to provide you with more help regarding the LoRa APRS project. I like this project very much.

Thanks again for your contribution!

73 de BD5HTY

peterus, 2021-03-31:
Sorry for the confusion! I had the feeling that you where a bot so i was very skeptical about it, I should have just looked up your callsign ;)
you are right! there is not a lot of information to get in contact with me :)
Please write me a email on oe5bpa@gmail.com and I will come back to you as soon as possible.
Thank you very much!

BD5HTY, 2021-04-01:
Copied, this post can be deleted, thank you very much!

dg0tm, 2021-07-15:
What is the purpose of it?

All kind of RX packets should passes through the iGate.

closed 2022-05-07 by peterus