lora-aprs/LoRa_APRS_iGate #77: Use WifiManager for configuration

SensorsIot, 2021-04-02:
I like your two projects a lot and was able to implement them! But I am quite familiar with ESP32 and PlatformIO. Not all HAMS are ESP32 cracks but still would like to use your project.
This is why I propose to use WiFiManager to enter the credentials, etc. This would enable to distribute only one bin file per board version. And maybe it would even be possible to automatically find out if it is a 0.7 or a 1.0 T-beam version by checking if the power management chip answers or not, for example.
Then you could use this simple project to distribute one file: https://github.com/4m1g0/Tiny-Uploader . It packs the bin file as well as the ESP32 uploader into one executable file (it is a fork of tasmotizer).
A lot of work for you, I know. But I think it would tremendously increase its reach.

peterus, 2021-05-14:
I like the concept of the WifiManager, but unfortunately I will not use it in my projects because of this reasons:
- WifiManager is using binary data instead of a structured file format (like json) in a file system (ftp would not work anymore)
- WifiManager is doing to much in my eyes (one lib should do one thing)

In the end I will go with a self build lib which can save structured data in a json file and display/edit the data in an http server. With this we have the same result and in the future many people just need to download the binary.
I am aware that the current solution is not the best as i am getting also a lot of mails asking for help.
Thanks for your feedback!

SensorsIot, 2021-05-14:
That is ok with me. I just thought such a concept is needed. Tasmota and Tasmotizer are a very good example and made it possible to distribute software in an efficient way as binaries.
Thank you for your effort!