lora-aprs/LoRa_APRS_iGate #81: Digipeater feature.

RadioMCMLVI, 2021-04-25:
Hallo, wäre es möglich, die optionale Funktion von LoRa Digipeater, die in den Versionen 2020 vorhanden war, wieder einzuführen?
Hi, would it be possible to re-introduce the optional feature of LoRa Digipeater that existed in the 2020 releases?
Robert
F1TZU

peterus, 2021-05-14:
Somebody is already working on it ;)

peterus, 2021-06-24:
merged with #96

closed 2021-06-24 by peterus

SensorsIot, 2021-06-25:
I upgraded to the newest release
Question: Is there a description of how to enable/use the digipeater function? I set digi active to true and aprs_is to false but I am not seeing any digipeating in the log of my "digipeater" igate (HB9BLA-21). Both igates receive the packet from a tracker. The internet-connected igate is HB9BLA-10

' with RSSI -62 and SNR 9.50
D            TaskRouter.cpp   56 : APRS-IS: disabled
D            TaskRouter.cpp   71 : DIGI: Source: HB9BLA-2, Destination: APLT00, Path: HB9BLA-21*, Type: Position Without Timestamp, Data: =4728.48N/00746.04E[000/000/A=001134LoRa Tracker, Info: github.com/lora-aprs/LoRa_APRS_Tracker - Bat.: 4.11V - Cur.: -111mA

D             TaskModem.cpp   42 : [00:05:19] Received packet 'Source: HB9BLA-2, Destination: APLT00, Path: , Type: Position Without Timestamp, Data: =4728.48N/00746.04E[000/000/A=001139LoRa Tracker, Info: github.com/lora-aprs/LoRa_APRS_Tracker - Bat.: 4.11V - Cur.: