lora-aprs/LoRa_APRS_iGate #84: Static IP für W-Lan

OE3BCB, 2021-05-02:
Ich würde Statische IPs benötigen.

// Set your Static IP address
IPAddress local_IP(192, 168, 1, 184);
// Set your Gateway IP address
IPAddress gateway(192, 168, 1, 1);

IPAddress subnet(255, 255, 0, 0);
IPAddress primaryDNS(8, 8, 8, 8);   //optional
IPAddress secondaryDNS(8, 8, 4, 4); //optional

Weitere Info: Der reconnect nach WLan verlust bleibt in einer Schleife hängen. Lässt sich nur nach Spannungslos machen beheben.

Danke für die viele Zeit für dieses Projekt.
73de Christoph

peterus, 2021-05-17:
don’t forget about Ethernet connection.

Hexathus, 2021-08-08:
Does the static ip function work?
Where and in what format should I fill it?

Thank you

dg0tm, 2021-08-18:
Hi,

yes it should! Add the following to is-cfg.json

"callsign": "NOCALL-10",

"network": {
         "DHCP": false,
         "staticIP": "192.0.2.100",
         "subnet": "255.255.255.0",
         "gateway": "192.0.2.1",
         "dns1": "192.0.2.1",
         "dns2": "192.0.2.2"
   },

     "wifi": {

73 de DG0TM