lora-aprs/LoRa_APRS_iGate #85: IGate does not use the coordinates from config file

2e0wkw, 2021-05-03:
**Describe the bug**
When the iGate beacons and I look it up in APRS.fi the position is off by several kilometres and the coordinates shown do not match those I entered into the config file.

**Expected behavior**
The location of the iGate should be represented as accurately as the owner desires so that users can calculate their distance from it.

**Screenshots**
In APRS.fi search for 2E0WKW and you will see that my iGate is KMs from my tracker even though they sit side by side on the shelf.

**Desktop (please complete the following information):**
 - OS: iOS 
 - PlatformIO Version: latest
 - Firmware Version: 21.14.0

**Board name:**
- TYGO LoRa v2

peterus, 2021-05-03:
@2e0wkw can you post your config/json file. thanks

2e0wkw, 2021-05-03:
Hi Peter, here it is:
"callsign": "2E0WKW-10",
    "wifi": {
        "AP": [
            {
                "SSID": "Garden-WIFI-2.4GHz",
		"password": “xxxxxxxx"
            }
        ]
    },
    "beacon": {
        "message": "LoRa iGATE for op 2E0WKW",
        "position": {
            "latitude": 51.129100,
            "longitude": -0.338400
        },
        "timeout": 15
    },
    "aprs_is": {
        "passcode": "15015",
        "server": "euro.aprs2.net",
        "port": 14580
    },
    "lora": {
        "frequency_rx": 433775000,
        "frequency_tx": 433775000,
        "power": 20,
        "spreading_factor": 12,
        "signal_bandwidth": 125000,
        "coding_rate4": 5
    },
    "display": {
        "always_on": true,
        "timeout": 10,
        "overwrite_pin": 0,
        "turn180": true
    },
    "ftp": {
        "active": false,
        "user": [
            {
                "name": "ftp",
                "password": "ftp"
            }
        ]
    },
    "ntp_server": "pool.ntp.org"
}

peterus, 2021-05-03:
I just checked you aprs.fi positions with the one from your config.
In the config you have a position which is close to Kingsfold, which looks good to 2E0WKW-10.
Your tracker 2E0WKW is close to Shalford.

So for me the position of the iGate is similar to the one from your config. Maybe check again your Lat and Long in your config.
Until now there is no issue known with the position reporting.

2e0wkw, 2021-05-03:
This is what APRS.fi is reporting for my iGate for me which is not what is in my config.

peterus, 2021-05-03:
sorry, i am not understanding correctly.
the lat and long from your config file are the same as in aprs.fi. so, there is no issue with the position reporting. maybe you have something wrong in the config with lat and long?

SensorsIot, 2021-05-03:
The position of your tracker seems to be (51.2152535, -0.5637836)

2e0wkw, 2021-05-04:
Hi, my confusion is that my tracker is showing up in the right place (Shalford) and so I copied the coordinates reported for it in APRS.if 51.1291 N and 0.3384 W and put them in the iGate config and yet it shows up 11 miles away.

My iOS APRS.Fi app (2E0WKW-5) also shows shows these same coordinates. So all three devices use the same coordinates but the iGate appears 11 miles away from my iPhone as the screen shot shows.

Thanks for bearing with me.

Eric

quarksteilchen, 2021-05-04:
Hi,
The difference between _51.2152535, -0.5637836_ and _51.1291 N, 0.3384 W_ is about 11.4 miles.
The second coordinate is in the middle of nowhere. This can't be right and should not go to any configuration file.
Your tracker shows "_51°12.91' N 0°33.84' W_" in aprs.fi

Now i see the problem.
_51°12.91' N 0°33.84' W_ does not equal _51.1291 N, 0.3384 W_.
These are two different formats of writing down coordinates.

You can use e.g. this website to convert between the most common formats: https://www.gps-coordinates.net/gps-coordinates-converter

2e0wkw, 2021-05-04:
Ah, ok thank you.

My apologies. User error and not a bug then.

Thanks all for the support in this matter.

Eric

closed 2021-05-04 by peterus