lora-aprs/LoRa_APRS_iGate #86: igated devices not seen at APRS.fi (RWA data)

emsieg, 2021-05-07:
Hi, 
when i check display, I do see incoming APRS messages from several ID´s.
So, decoding is working, but then I check, which ID`s have been forwarded to APRS, I cannot see a single one:

e.g, 07-May 00:00 till 08:27
2021-05-07 00:00:42 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 00:05:40 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 00:10:40 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 00:15:40 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 00:20:40 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 00:25:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 00:30:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 00:35:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 00:40:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 00:45:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 00:50:42 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 00:55:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 01:00:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 01:05:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2NUERNBG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 01:10:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 01:15:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 01:20:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 01:25:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 01:30:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 01:35:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 01:40:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 01:45:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 01:50:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 01:55:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 02:00:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 02:05:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 02:10:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 02:15:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 02:20:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 02:25:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 02:30:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 02:35:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 02:40:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 02:45:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 02:50:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 02:55:43 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 03:00:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 03:05:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FFL:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 03:10:42 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 03:15:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 03:20:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 03:25:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 03:30:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 03:35:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 03:40:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 03:45:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 03:50:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 03:55:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 04:00:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 04:05:43 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 04:10:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 04:15:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 04:20:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 04:25:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2KA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 04:30:42 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2LEIPZIG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 04:35:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2LEIPZIG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 04:40:42 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2LEIPZIG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 04:45:42 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2LEIPZIG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 04:50:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2LEIPZIG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 04:55:42 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2LEIPZIG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 05:00:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2LEIPZIG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 05:05:41 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2LEIPZIG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 05:10:42 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2LEIPZIG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 05:15:43 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2LEIPZIG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 05:20:43 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2LEIPZIG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
[... 33 lines cut ...]
2021-05-07 08:10:42 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2LEIPZIG:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 08:12:56 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FRA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 08:17:57 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FRA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 08:22:56 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FRA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
2021-05-07 08:27:57 PDT: DL6SIX-10>APLG01,TCPIP*,qAC,T2FRA:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E

I would expect to see something like (example from another LoRa device:
2021-05-07 07:31:59 PDT: DJ7OO-14>APRS,qAS,**DF9PM-5**:!L56m^PTV?&EiQ LoRa-DiGi MZ-Lbg. SNR=+8dB RSSI=-76db
2021-05-07 07:41:57 PDT: DJ7OO-14>APRS,qAS,**DL0MZ-16**:!L56m^PTV?&EiQ LoRa-DiGi MZ-Lbg.
2021-05-07 07:41:57 PDT: DJ7OO-14>APRS,qAS,**DF9PM-5**:!L56m^PTV?&EiQ LoRa-DiGi MZ-Lbg. SNR=+8dB RSSI=-75db
2021-05-07 07:51:56 PDT: DJ7OO-14>APRS,qAS,**DL0MZ-16**:!L56m^PTV?&EiQ LoRa-DiGi MZ-Lbg.
2021-05-07 07:51:56 PDT: DJ7OO-14>APRS,qAS,**DF9PM-5**:!L56m^PTV?&EiQ LoRa-DiGi MZ-Lbg. SNR=+8dB RSSI=-74db
2021-05-07 08:01:54 PDT: DJ7OO-14>APRS,qAS,**DL0MZ-16**:!L56m^PTV?&EiQ LoRa-DiGi MZ-Lbg.
[... cut ...]
2021-05-07 08:11:52 PDT: DJ7OO-14>APRS,qAS,**DL0MZ-16**:!L56m^PTV?&EiQ LoRa-DiGi MZ-Lbg.
2021-05-07 08:11:52 PDT: DJ7OO-14>APRS,qAS,**DF9PM-5**:!L56m^PTV?&EiQ LoRa-DiGi MZ-Lbg. SNR=+8dB RSSI=-72db
2021-05-07 08:21:51 PDT: DJ7OO-14>APRS,qAS,**DL0MZ-16**:!L56m^PTV?&EiQ LoRa-DiGi MZ-Lbg.
2021-05-07 08:21:51 PDT: DJ7OO-14>APRS,qAS,**DF9PM**-5**:!L56m^PTV?&EiQ LoRa-DiGi MZ-Lbg. SNR=+8dB RSSI=-77db
2021-05-07 08:21:56 PDT: DJ7OO-14>APRS,**DJ8HS-30***,qAS,**DK1KT-13:**!L56m^PTV?&EiQ LoRa-DiGi MZ-Lbg. SNR=-16dB RSSI=-86db

I´ve installed latest version.

regards
Gustav

peterus, 2021-05-07:
please check the following:
- start the serial monitor
- reset the board
- you will see information from the server if the connection was successful or if there is no connection to the server or if the passcode is wrong

with this information we can further debug.

emsieg, 2021-05-07:
Hi,
igate is connected to APRS Server, and I do see my LoRa Tracker in APRS.fi correctly and locations get updated.
but i don´t see them in the raw data, which is confusing.
Monitoring the display for a longer timeframe, i see some other LoRa devices as well.

I tried to see, which i was igating to APRS Server and here i saw the (kind of) issue.

here is the log, from monitor

␛[0m␛[0;33mW       LoRa_APRS_iGate.cpp   35 : Version: 21.14.0
␛[0m␛[0;32mI       LoRa_APRS_iGate.cpp   65 : Board ␛[0m␛[0;32mTTGO_LORA32_V2␛[0m␛[0;32m loaded.
␛[0m␛[0;33mW           TaskManager.cpp   33 : call setup from ␛[0m␛[0;33mWifiTask
␛[0m␛[0;34mD              TaskWifi.cpp   20 : Looking for AP: ␛[0m␛[0;34mFuerstberg
␛[0m␛[0;34mD              TaskWifi.cpp   20 : Looking for AP: ␛[0m␛[0;34mFRITZ!Box 6660C WP2
␛[0m␛[0;34mD              TaskWifi.cpp   20 : Looking for AP: ␛[0m␛[0;34miPhone8 now
␛[0m␛[0;34mD              TaskWifi.cpp   20 : Looking for AP: ␛[0m␛[0;34mF39
␛[0m␛[0;33mW           TaskManager.cpp   38 : call setup from ␛[0m␛[0;33mDisplayTask
␛[0m␛[0;33mW           TaskManager.cpp   38 : call setup from ␛[0m␛[0;33mLoraTask
␛[0m␛[0;33mW           TaskManager.cpp   38 : call setup from ␛[0m␛[0;33mOTATask
␛[0m␛[0;33mW           TaskManager.cpp   38 : call setup from ␛[0m␛[0;33mNTPTask
␛[0m␛[0;33mW           TaskManager.cpp   38 : call setup from ␛[0m␛[0;33mAprsIsTask
␛[0m␛[0;32mI       LoRa_APRS_iGate.cpp  114 : setup done...
␛[0m␛[0;34mD              TaskWifi.cpp   37 : IP address: ␛[0m␛[0;34m192.168.2.42
␛[0m␛[0;32mI               TaskNTP.cpp   29 : Current time: ␛[0m␛[0;32m15:51:47
␛[0m␛[0;32mI            TaskAprsIs.cpp   67 : connecting to APRS-IS server: ␛[0m␛[0;32mgermany.aprs2.net␛[0m␛[0;32m on port: ␛[0m␛[0;32m14580
␛[0m␛[0;32mI            TaskAprsIs.cpp   75 : Connected to APRS-IS server!
␛[0m␛[0;34mD            TaskAprsIs.cpp   54 : [15:51:47] ␛[0m␛[0;34mDL6SIX-10>APLG01:=4946.10NL00838.15E&LoRa iGATE, by DL6SIX/KE6E
␛[0m␛[0;34mD .pio\libdeps\lora_board\APRS-IS-Lib\src\APRS-IS.cpp   97 : # aprsc 2.1.5-g8af3cdc 7 May 2021 15:52:06 GMT T2FFL 44.149.16.132:14580
␛[0m␛[0;34mD .pio\libdeps\lora_board\APRS-IS-Lib\src\APRS-IS.cpp   97 : # aprsc 2.1.5-g8af3cdc 7 May 2021 15:52:26 GMT T2FFL 44.149.16.132:14580
␛[0m␛[0;34mD              TaskLora.cpp   40 : [15:52:29] ␛[0m␛[0;34mReceived packet '␛[0m␛[0;34mSource: DL6SIX-11, Destination: APRS, Path: , Type: Position Without Timestamp, Data: !/5?ZiP^Xd[z3H/A=000557 Batt=3.70VTTGO-APRS Tracker & WX, by DL6SIX/KE6E␛[0m␛[0;34m' with RSSI ␛[0m␛[0;34m-28␛[0m␛[0;34m and SNR ␛[0m␛[0;34m9.00
␛[0m␛[0;34mD .pio\libdeps\lora_board\APRS-IS-Lib\src\APRS-IS.cpp   97 : # aprsc 2.1.5-g8af3cdc 7 May 2021 15:52:46 GMT T2FFL 44.149.16.132:14580
␛[0m␛[0;32mI               TaskNTP.cpp   29 : Current time: ␛[0m␛[0;32m15:52:47
␛[0m␛[0;34mD .pio\libdeps\lora_board\APRS-IS-Lib\src\APRS-IS.cpp   97 : # aprsc 2.1.5-g8af3cdc 7 May 2021 15:53:06 GMT T2FFL 44.149.16.132:14580
␛[0m␛[0;34mD .pio\libdeps\lora_board\APRS-IS-Lib\src\APRS-IS.cpp   97 : # aprsc 2.1.5-g8af3cdc 7 May 2021 15:53:26 GMT T2FFL 44.149.16.132:14580
␛[0m␛[0;34mD              TaskLora.cpp   40 : [15:53:31] ␛[0m␛[0;34mReceived packet '␛[0m␛[0;34mSource: DL6SIX-11, Destination: APRS, Path: , Type: Position Without Timestamp, Data: !/5?\mP^Z0_z,H.../...g...t090r...p...P...h15b.....TTGO-APRS Tracker & WX, by DL6SIX/KE6E␛[0m␛[0;34m' with RSSI ␛[0m␛[0;34m-17␛[0m␛[0;34m and SNR ␛[0m␛[0;34m12.75
␛[0m␛[0;34mD .pio\libdeps\lora_board\APRS-IS-Lib\src\APRS-IS.cpp   97 : # aprsc 2.1.5-g8af3cdc 7 May 2021 15:53:46 GMT T2FFL 44.149.16.132:14580
␛[0m␛[0;32mI               TaskNTP.cpp   29 : Current time: ␛[0m␛[0;32m15:53:47
␛[0m␛[0;34mD              TaskLora.cpp   40 : [15:54:00] ␛[0m␛[0;34mReceived packet '␛[0m␛[0;34mSource: DL6SIX-11, Destination: APRS, Path: , Type: Position Without Timestamp, Data: !/5?\sP^ZL[;.H/A=000444 Batt=3.71VTTGO-APRS Tracker & WX, by DL6SIX/KE6E␛[0m␛[0;34m' with RSSI ␛[0m␛[0;34m-19␛[0m␛[0;34m and SNR ␛[0m␛[0;34m10.75
␛[0m␛[0;34mD .pio\libdeps\lora_board\APRS-IS-Lib\src\APRS-IS.cpp   97 : # aprsc 2.1.5-g8af3cdc 7 May 2021 15:54:06 GMT T2FFL 44.149.16.132:14580
␛[0m␛[0;34mD              TaskLora.cpp   40 : [15:54:10] ␛[0m␛[0;34mReceived packet '␛[0m␛[0;34mSource: DL6SIX-11, Destination: APRS, Path: , Type: Position Without Timestamp, Data: !/5?\IP^Yf_s2H.../...g...t090r...p...P...h15b.....TTGO-APRS Tracker & WX, by DL6SIX/KE6E␛[0m␛[0;34m' with RSSI ␛[0m␛[0;34m-18␛[0m␛[0;34m and SNR ␛[0m␛[0;34m10.00
␛[0m␛[0;34mD              TaskLora.cpp   40 : [15:54:17] ␛[0m␛[0;34mReceived packet '␛[0m␛[0;34mSource: DL6SIX-11, Destination: APRS, Path: , Type: Position Without Timestamp, Data: !/5?\9P^YC[k.H/A=000409 Batt=3.71VTTGO-APRS Tracker & WX, by DL6SIX/KE6E␛[0m␛[0;34m' with RSSI ␛[0m␛[0;34m-19␛[0m␛[0;34m and SNR ␛[0m␛[0;34m8.75
␛[0m␛[0;34mD .pio\libdeps\lora_board\APRS-IS-Lib\src\APRS-IS.cpp   97 : # aprsc 2.1.5-g8af3cdc 7 May 2021 15:54:26 GMT T2FFL 44.149.16.132:14580
␛[0m␛[0;34mD .pio\libdeps\lora_board\APRS-IS-Lib\src\APRS-IS.cpp   97 : # aprsc 2.1.5-g8af3cdc 7 May 2021 15:54:46 GMT T2FFL 44.149.16.132:14580
␛[0m␛[0;32mI               TaskNTP.cpp   29 : Current time: ␛[0m␛[0;32m15:54:47
␛[0m␛[0;34mD .pio\libdeps\lora_board\APRS-IS-Lib\src\APRS-IS.cpp   97 : # aprsc 2.1.5-g8af3cdc 7 May 2021 15:55:06 GMT T2FFL 44.149.16.132:14580
[... cut ...]
␛[0m␛[0;34mD              TaskLora.cpp   40 : [15:59:41] ␛[0m␛[0;34mReceived packet '␛[0m␛[0;34mSource: DL6SIX-11, Destination: APRS, Path: , Type: Position Without Timestamp, Data: !/5?\mP^Y\_>7H.../...g...t091r...p...P...h20b.....TTGO-APRS Tracker & WX, by DL6SIX/KE6E␛[0m␛[0;34m' with RSSI ␛[0m␛[0;34m-26␛[0m␛[0;34m and SNR ␛[0m␛[0;34m9.50
␛[0m␛[0;34mD .pio\libdeps\lora_board\APRS-IS-Lib\src\APRS-IS.cpp   97 : # aprsc 2.1.5-g8af3cdc 7 May 2021 15:59:46 GMT T2FFL 44.149.16.132:14580
␛[0m␛[0;32mI               TaskNTP.cpp   29 : Current time: ␛[0m␛[0;32m15:59:47

my igate is: DL6SIX-10
my Tracker (& WX) is DL6SIX-11

peterus, 2021-05-07:
so, everything is looking good to me.
all packages are forwarded without a problem and there is no aprs-is issue on the log.

is it possible that a different igate is close to you? because if another igate is faster than your igate to deliver the package to the aprs-is server the other package will be used and your package is just droped.
this is done because the aprs-is servers should not spammed with the same packages from different servers.

K4KDR, 2021-05-08:
Hello!  Please forgive if I am not understanding the question or the logs correctly.

However, if my understanding is correct, the APRS-IS network logs packets forwarded from an iGate received over RF, not the local network.  Please note in your original post that the "digipeater" (station ID with the * flag) is listed as "TCPIP*".   

That would indicate that those packets were received over a LAN connection and not over-the-air.

-Scott,  K4KDR

closed 2021-05-14 by peterus