lora-aprs/LoRa_APRS_iGate #88: No reconnection to WLAN after lost WLAN connection

Jup88, 2021-05-13:
When WLAN connection to the iGate gets lost, the iGate doesn't do an automatic re-connection to the WLAN. It stops working and the screen says " booting". 

booting screen:
[image]

Automatic restart including notification to WLAN would be very appreciated to ensure an 24/7 operation.

Firmware Version: Release v21.14.0

Board: TTGO-t-beam-v1

Peter, it would be great if you could get it fixed. Your work is highly appreciated. We all like your iGate solution :-)

peterus, 2021-05-13:
Thanks for the ticket.
There are currently a lot of issues with Wifi which i will fix asap!

peterus, 2021-06-24:
there are a lot of improvements in the last release #96. please check and report again if there are issues :)

closed 2021-06-24 by peterus

Jup88, 2021-06-25:
Hi Peter,

many, many thanks for updating the iGate Firmware to improve the Wifi
reconnection issues. I have done the update and had no issues so far. I
will give you feedback the next days how it works.

Highly appreciate your pioneers work.

Wish you all the best.

73 de Klaus, DL1ND

Jup88, 2021-07-29:
Hi Peter,

I would like to give you the feedback that WLAN is working very stable now.
In my case 4 weeks without any issues so far.

Thanks for your work.

73 de Klaus DL1ND

reinald100, 2022-08-13:
Hallo ich bekomme keine Verbindung zum Aprsis  Server.Es steht immer Aprsis not connected.Muss ich in der Fritz Box noch einen Port eventuell freischalten? 
Die WLAN Verbindung geht ohne Probleme.