lora-aprs/LoRa_APRS_iGate #92: link to vhf/uhf

icicmoi, 2021-06-02:
A gateway to Direwolf (or other, ie kiss access) would really open up to the world when the internet -igate- remains unavailable.
In other words: allow trackers to get out of this LoRa bubble ... by a radio solution and thus become a real complement of coverage for "APRS 144.800 MHz -and gsm- white areas".
... and because the work done is already so great :-)
73 '   Philippe, f5rvx

peterus, 2021-06-03:
Hi Philippe!
I was already thinking a lot about different solutions to support other protocols etc.
but unfortunately i want to have this firmware as simple as possible. so i will not include kiss in this project.
but there is a solution for this: https://github.com/M0IGA/ESP32_KISS_LoRa_APRS_1200
73 Peter, OE5BPA

closed 2021-06-03 by peterus

icicmoi, 2021-06-03:
Hi Peter and thank you very much for your prompt response :-)

1- I completely understand your search for simplicity ... and it is also successful: I am the perfect example of the eternal beginner in computer science ... and I got out without problem to install tracker and igate ;)
_(to be honest just needed some help to adapt the code to a ttgo "paxcounter" less expensive than the Tbeam + gps dropped from the sky (Trimble copernicus from M10 radiosonde.)_ 

2- Thank you very much also for the link to the M0IGA github: I did not know this work and will quickly experiment :-)  

3- To finish and since it is easier to propose than to code (!) I will add: the ideal would be that the different LoRa trackers speak the same dialect to be understood by one or the other of the digipeaters, igates, kiss-tnc ...  

Best 73's.
Philippe, f5rvx