lora-aprs/LoRa_APRS_iGate #93: LoRa_APRS_iGate.cpp  102 : You have to change your settings in 'data/is-cfg.json' and upload it via "Upload File System image"!

antonjan, 2021-06-03:
HI
Dont think this is a bug but I get this error or startup after flash.
It seems it is not seeing the config or I might have it wrongly configured.
I am using a TTGO-T-Beam board.
Here is the log on startup.
***************** start *****************
Using '/dev/ttyUSB0' as serial port.
Connecting...[07:53:25]ets Jul 29 2019 12:21:46
[07:53:25]
[07:53:25]rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
[07:53:25]waiting for download
.
Serial port closed!
.
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 24:0A:C4:F7:D9:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.6s
Compressed 15872 bytes to 10319...
Writing at 0x00001000... (100 %)Wrote 15872 bytes (10319 compressed) at 0x00001000 in 0.2 seconds (effective 519.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...
Writing at 0x00008000... (100 %)Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.0 seconds (effective 4585.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 21232.7 kbit/s)...
Hash of data verified.
Compressed 919104 bytes to 515898...
Writing at 0x0008c000... (100 %)Wrote 919104 bytes (515898 compressed) at 0x00010000 in 12.2 seconds (effective 604.6 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[07:53:44]W       LoRa_APRS_iGate.cpp   34 : LoRa APRS iGate by OE5BPA (Peter Buchegger)
[07:53:44]W       LoRa_APRS_iGate.cpp   35 : Version: 21.14.0
[07:53:44]E (561) SPIFFS: mount failed, -10025
[07:53:45]W         configuration.cpp   27 : Failed to read file, using default configuration.
[07:53:45]I           BoardFinder.cpp   13 : looking for a board config.
[07:53:45]I           BoardFinder.cpp   14 : searching for OLED...
[07:53:45]D           BoardFinder.cpp  128 : 3
[07:53:45]D           BoardFinder.cpp  130 : power chip found!
[07:53:45]D           BoardFinder.cpp  128 : 3
[07:53:45]D           BoardFinder.cpp  130 : power chip found!
[07:53:45]D           BoardFinder.cpp  128 : 3
[07:53:45]D           BoardFinder.cpp  130 : power chip found!
[07:53:45][E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=0, scl=0
[07:53:45]W           BoardFinder.cpp   76 : issue with wire
[07:53:45]W           BoardFinder.cpp   33 : could not find OLED, will search for the modem now...
[07:53:45]I           BoardFinder.cpp   44 : found a board config: TTGO_LORA32_V1
[07:53:45]I       LoRa_APRS_iGate.cpp   62 : will restart board now!
[07:53:45]ets Jul 29 2019 12:21:46
[07:53:45]
[07:53:45]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[07:53:45]configsip: 0, SPIWP:0xee
[07:53:45]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[07:53:45]mode:DIO, clock div:2
[07:53:45]load:0x3fff0018,len:4
[07:53:45]load:0x3fff001c,len:1044
[07:53:45]load:0x40078000,len:8896
[07:53:45]load:0x40080400,len:5828
[07:53:45]entry 0x400806ac
[07:53:46]W       LoRa_APRS_iGate.cpp   34 : LoRa APRS iGate by OE5BPA (Peter Buchegger)
[07:53:46]W       LoRa_APRS_iGate.cpp   35 : Version: 21.14.0
[07:53:46]I       LoRa_APRS_iGate.cpp   65 : Board TTGO_LORA32_V1 loaded.
[07:53:46]W           TaskManager.cpp   33 : call setup from WifiTask
[07:53:46]W           TaskManager.cpp   38 : call setup from DisplayTask
[07:53:46]W           TaskManager.cpp   38 : call setup from LoraTask
[07:53:46]W           TaskManager.cpp   38 : call setup from OTATask
[07:53:46]W           TaskManager.cpp   38 : call setup from NTPTask
[07:53:46]W           TaskManager.cpp   38 : call setup from AprsIsTask
[07:53:46]E       LoRa_APRS_iGate.cpp  102 : You have to change your settings in 'data/is-cfg.json' and upload it via "Upload File System image"!
******************************** end ******************************
Here is my config file is-cfg.json
********************** start ***************************************
{
	"callsign": "ZR6AIC-11",
	"wifi": {
		"AP": [
			{
				"active": true,
				"SSID": "Giga8",
				"password": "6409125086089"
			}
		]
	},
	"beacon": {
		"message": "LoRa iGATE & Digi, Info: github.com/antonjan/LoRa_APRS_iGate",
		"position": {
			"latitude": -26.2854385,
			"longitude": 28.0713523
		},
		"timeout": 15
	},
	"aprs_is": {
		"active": true,
		"passcode": "22194",
		"server": "euro.aprs2.net",
		"port": 14580
	},
	"lora": {
		"frequency_rx": 433775000,
		"frequency_tx": 433775000,
		"power": 20,
		"spreading_factor": 12,
		"signal_bandwidth": 125000,
		"coding_rate4": 5
	},
	"display": {
		"always_on": true,
		"timeout": 10,
		"overwrite_pin": 0,
		"turn180": true
	},
	"digi": {
		"active": true,
		"forward_tineout": 5,
		"beacon": true,
		"beacon_timeout": 30
	},
	"ftp": {
		"active": false,
		"user": [
			{
				"name": "ftp",
				"password": "ftp"
			}
		]
	},
	"ntp_server": "pool.ntp.org"
}
******************************** stop ******************************************
Here is also my build file platformio.ini
********************************* start *****************************************
[env:lora_board]
board = esp32doit-devkit-v1
#board = ttgo-t-beam-v1
build_flags = -Werror -Wall -DNO_GLOBAL_INSTANCES

[env:lora_board_debug]
board = esp32doit-devkit-v1
build_flags = -Werror -Wall -DNO_GLOBAL_INSTANCES
build_type = debug
monitor_filters = esp32_exception_decoder

I have tryded to change the board to ttgo-t-beam but then Board is not found

Hope you can assist
Regards
Anton 
ZR6AIC

antonjan, 2021-06-03:
I found the problem.
I had to add the following in the platformio.ini
upload_port = /dev/ttyUSB*  in the ini file and then it uploaded the is-cfg.json config file

peterus, 2021-06-03:
Hi Anton!
looks like platformio could not found the correct port in your case to upload the firmware and/or the configuration/file system.
thank you for sharing your fix!
73 Peter, OE5BPA

closed 2021-06-03 by peterus

gilrpinheiro, 2022-01-27:
I am having the same issue! The firmware is being loaded in my TTGO-T-Beam board, but its OLED display show the error message: You have to change your settings in "data/is-cfg.json" and upload via "UploadFile System Image"!

oh2lak, 2022-01-28:
Have you followed the instructions and uploaded the settings file to the
board?

This is a separate task to be done after the board is being flashed

73,
Erik OH2LAK