lora-aprs/LoRa_APRS_iGate #95: activate internal watchdog to reset board if it is hanging

peterus, 2021-06-23:
example code: https://www.az-delivery.de/blogs/azdelivery-blog-fur-arduino-und-raspberry-pi/watchdog-und-heartbeat

iu2frl, 2021-11-28:
Maybe this can help: https://iotespresso.com/create-custom-watchdog-timer-in-esp32/

RoWaBe, 2021-12-10:
Yes yes yes, very good idea - my iGate for example hangs all 3...7 days. Unfortunately I'm not able to find out why. For those cases a watchdog is a good solution. Momently have a workaroud unsing an external power supply timer to restet once each day in the night.

Mane76, 2022-01-11:
yes, very good idea. My iGate hangs after ~20h, also having not been able to find out why. 
I am available also for beta-testing.

Mane76, 2022-02-05:
after excessive testing I found out that with a poor wifi connection the iGate hangs after a certain period of time, with a good wifi connection no hangings occur.

iu2frl, 2022-02-05:
> after excessive testing I found out that with a poor wifi connection the iGate hangs after a certain period of time, with a good wifi connection no hangings occur.

Totally aggree, problem is that my connection drops from time to time or after a reboot of internet modem, every time this happens i have to reboot LoRa too or it gets stuck somewhere

closed 2022-11-16 by peterus