lora-aprs/LoRa_APRS_iGate #98: Software hangs if

dl9rdz, 2021-07-11:
**Describe the bug**
If "digi"->"beacon" is set to true in the config, the software hangs shortly after startup.

**To Reproduce**
Steps to reproduce the behaviour:
- Set beacon to true in digi section in data/is-cfg.json, upload the config (pio run --target=uploadfs)

**Expected behavior**
I expect the board to run normally

**Screenshots**
Log from console (after the last line, no more ouput, completely stuck):
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
I       LoRa_APRS_iGate.cpp   47 : LoRa APRS iGate by OE5BPA (Peter Buchegger)
I       LoRa_APRS_iGate.cpp   48 : Version: 21.25.0
I       LoRa_APRS_iGate.cpp   79 : Board TTGO_LORA32_V2 loaded.
D           TaskManager.cpp   23 : call setup from WifiTask
D              TaskWifi.cpp   19 : Looking for AP: Dinosauro
[I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Dinosauro
lD           TaskManager.cpp   28 : call setup from ModemTask
D           TaskManager.cpp   28 : call setup from RouterTask
D           TaskManager.cpp   28 : call setup from OTATask
D           TaskManager.cpp   28 : call setup from NTPTask
D           TaskManager.cpp   28 : call setup from AprsIsTask
I       LoRa_APRS_iGate.cpp  139 : setup done...
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
I               TaskEth.cpp   13 : WiFi Started
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[I][WiFiMulti.cpp:119] run(): [WIFI] 4 networks found
[D][WiFiMulti.cpp:151] run():        0: [1][2E:3A:FD:9D:06:87] FRITZ!Box Fon WLAN 7270 (-50) *
[D][WiFiMulti.cpp:149] run():  --->   1: [1][2C:3A:FD:9D:06:87] Dinosauro (-51) *
[D][WiFiMulti.cpp:151] run():        2: [1][62:01:94:B9:45:24] LEDnetB94524 (-75)  
[D][WiFiMulti.cpp:151] run():        3: [11][98:DA:C4:27:FD:65] WLAN-947009_EXT (-86) *
[I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 2C:3A:FD:9D:06:87 SSID: Dinosauro Channel: 1 (-51)
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
I               TaskEth.cpp   16 : WiFi Connected
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.42.23, MASK: 255.255.255.0, GW: 192.168.42.1
I               TaskEth.cpp   19 : WiFi MAC: D8:A0:1D:60:D3:BC, IPv4: 192.168.42.23
[I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
[D][WiFiMulti.cpp:175] run(): [WIFI] SSID: Dinosauro
[D][WiFiMulti.cpp:176] run(): [WIFI] IP: 192.168.42.23
[D][WiFiMulti.cpp:177] run(): [WIFI] MAC: 2C:3A:FD:9D:06:87
[D][WiFiMulti.cpp:178] run(): [WIFI] Channel: 1
D              TaskWifi.cpp   36 : IP address: 192.168.42.23
D            TaskRouter.cpp   81 : [00:00:08] DL9RDZ-10>APLG01:=4835.14NL01326.24E&LoRa iGATE & Digi DL9RDZ
[I][ArduinoOTA.cpp:155] begin(): OTA server at: DL9RDZ-10.local:3232
I               TaskNTP.cpp   30 : Current time: 11:47:47
I            TaskAprsIs.cpp   44 : connecting to APRS-IS server: euro.aprs2.net on port: 14580
I            TaskAprsIs.cpp   52 : Connected to APRS-IS server!
```

OLED Display (stuck there)
```
Display:DL9RDZ-10
Modem:Okay
Router:Booting
OTA:Okay
NTP:Booting
AprsIs:Booting
```

**Desktop (please complete the following information):**
 - OS: MacOS
 - PlatformIO Version: platformio/contrib-piohome 3.3.4, platform: espressif32 @ 3.1.1
 - Firmware Version: current from git, tagged v21.25.0

**Board name:**
- TTGO LoRa32 V2.1 (printed on board: T3_V1.6 20180606)

**Additional context**
When changing "true" to "false" in the config, everything runs as expected (i.e., Display shows AprsIs: Connected, NTP: time, etc, and serial port monitor prints lines (APRS-IS:..., TaskNTP...) after "Connected to APRS-IS server".

dl9rdz, 2021-07-11:
Seems to be a hardware issue.

In LoRa.cpp "wait for TX done" was stuck in an infinite loop. Tried a different (identical) board, and the problem is gone...

dg0tm, 2021-07-15:
https://github.com/sandeepmistry/arduino-LoRa/issues/378

peterus, 2021-07-15:
@dg0tm as it is the first time somebody reported something like this i will close the ticket now. in the future we can think about a retry.

closed 2021-07-15 by peterus