lora-aprs/LoRa_APRS_iGate #111: wakeup OLED with button not implemented

peterus, 2021-09-03:
see https://github.com/lora-aprs/LoRa_APRS_iGate/discussions/110

OK1FWG, 2021-09-09:
I am using the devel branch where some changes were implemented and I can confirm, that this is working. But!
In the beginning of my testing, I was surprised, that I must have the override pin connected either to GND or 3V3 (depending on the logical condition). I then tried several different GPIOs (with or without PULLUP). This was not what I wanted - simply connect the overridepin to GND should wake up the display. But this wasn't working. I choosed the GPIO13 (has the PULLUP) and had to change this in LoRa_APRS_iGate.cpp:
from:
```
  if (userConfig.display.overwritePin != 0) {
    pinMode(userConfig.display.overwritePin, INPUT);
    pinMode(userConfig.display.overwritePin, INPUT_PULLUP);
  }
 ```
to:
```
if (userConfig.display.overwritePin != 0) {
    pinMode(userConfig.display.overwritePin, INPUT_PULLUP);
  }
```

I understand the doubled pin input mode settings (you may use different pin than me), but I really wonder why it caused troubles. So my observation. Now I am satisfied ;-)

peterus, 2021-12-18:
closing with v21.50.0.

closed 2021-12-18 by peterus