lora-aprs/LoRa_APRS_iGate #112: TTGO T-Beam loosing configuration

iu2frl, 2021-09-05:
**Describe the bug**
Using software version 21.35.0 on a TTGO T-Beam woftware seems to hang and overflow overwriting some functional parameters such as config info

**To Reproduce**
Steps to reproduce the behavior:
1. build and upload software
2. Edit callsign and coordinates on "is-cfg.json"
3. click on "build filesystem image" and "upload file system image"
4. wait some hours of normal work (usually from 8 to 15) and the software will hang (OLED stuck on any received package)., restarting the TTGO will drive to the "Error: you have to change settings..." page on the OLED. Only way to recover is to upload file system image once more

**Screenshots**
Check packet list of "IU2FRL-10" on aprs.fi to see the boot and when it hangs.

**Board name:**
- TTGO Lora32 T-Beam with OLED 433MHz
- Powered with internal 18650 battery and external power bank on USB

**Additional context**
First try with "digi beacon: true" lasted around 8h, with "digi beacon: false" lasted around 24h and then i had to upload again. I assume some function is overflow and editing some required info

iu2frl, 2021-09-05:
By reading other post the behaviour seems somehow related to #69 or at least same result. I tend to exclude power problems as i'm using both internal battery and external power bank

iu2frl, 2021-09-18:
Further investigations revealed that problem appears in any board when trying to power on with low voltage battery, seems that is related to filesystem corruption

iu2frl, 2021-09-22:
Problem also appears in other LoRa boards. When battery drops configuration is erased/damaged and needs new flash. Please find different way to save config, maybe EEPROM?

peterus, 2022-05-07:
I will remove the re-write of the configuration after a start-up.

closed 2022-05-20 by peterus