lora-aprs/LoRa_APRS_iGate #114: No RX with 21.35.0-develop

DB2FM, 2021-09-14:
With the newest develop-Branch of today my iGate didn't receive anymore. 
Wet back to master and everything works as expected. 

**Board name:**
- PoE-iGate

peterus, 2021-09-14:
thanks for this ticket! this information is very important! i will take a look into this 😉

peterus, 2021-09-19:
@DB2FM i reverted a big change, please re-test and revert. thanks!

DB2FM, 2021-09-20:
@peterus Hi Peter,

I don’t see develop branch anymore. There are only three branches: master, http_config and python_test_framework.
Which one should I try?

peterus, 2021-09-20:
@DB2FM by accident i deleted the branch yesterday. i could revert the deletion and have setup a protection for the branch.
please try again with the developer branch.

DB2FM, 2021-09-20:
@peterus Hi Peter,
When I download ZIP in branch develop I get master.zip of 01.09.2021...

peterus, 2021-09-20:
@DB2FM just checked it: the download zip is always pointing to the master branch for me even if i change the branch to another one.

DB2FM, 2021-09-20:
@peterus Same for me.

peterus, 2021-09-20:
```
git clone https://github.com/lora-aprs/LoRa_APRS_iGate.git
git checkout develop
```

DB2FM, 2021-09-20:
RX does work again :)

> Am 20.09.2021 um 07:48 schrieb Peter Buchegger ***@***.***>:
> 
> git checkout develop

closed 2022-05-07 by peterus