lora-aprs/LoRa_APRS_iGate #121: Add parseable diagnostic data output avilable via HTTP or Raw-TCP socket

dm5tt, 2021-10-05:
**Is your feature request related to a problem? Please describe.**

It would be great to have some kind of diagnostics output available which allows us to collect statistic data of our LoRA iGates.

Using this data we can analyze/improve and visualize (Grafana?) our setups  using TSDB (Time Series Databases).

**Describe the solution you'd like**

Basically the setup will look like this:

[LoRa iGate] <--- Raw TCP Socket or  HTTP --- [Python Collector Script] ---> InFlux DB ---> Grafana

Example (with random data) of a diagnostic data payload:

```
{
  "esp32_unique_id": "FA3CFFAEFAAA"
  "callsign": "dm5tt-2",
   "timestamp": "2017-10-14T22:11:20+0000",
  "coordinates": {
    "lat": "42.111",
    "lon": "9.111",
    "alt": "191"
  },
  "lora": {
    "rssi": "-96",
    "snr": "-100",
    "payload_b64": "VEhJU0lTQVRFU1QxMjM1NlRISVNJU0FURVNUMTIzNTZUSElTSVNBVEVTVDEyMzU2VEhJU0lTQVRFU1QxMjM1NlRISVNJU0FURVNUMTIzNTZUSElTSVNBVEVTVDEyMzU2VEhJU0lTQVRFU1QxMjM1NlRISVNJU0FURVNUMTIzNTZUSElTSVNBVEVTVDEyMzU2VEhJU0lTQVRFU1QxMjM1Ng=="
  }
}
```

peterus, 2021-10-05:
This is definitely possible and if I think more about a very nice idea ;)
but I would try another approach: write directly to the InFluxDB ;)

I found this library which could work: https://github.com/teebr/Influx-Arduino

Will test more and report back with something ;)

SensorsIot, 2021-10-05:
In IOT a lot of people use the MQTT protocol which is supported by most devices. This would make it very flexible, and you find an MQTT library for the ESP32, too.

TechMindsYT, 2022-01-02:
At the very least an HTTP server on the iGate with a status page would be awesome!