lora-aprs/LoRa_APRS_iGate #153: Starting Lora Failed

iw9grl, 2022-01-26:
**Describe the bug**
Hi
me and a friend of mine bought the board : Wemos LILYGO TTGO LORA SX1278 ESP32

the board unfortunately didn't have an UART connection but anyway we solved this issue with a workaround using an extern UART-TTL converter.

We are able to upload the firmware correctly but we receive this message : 

[image]

It seems that SX1278 is not available, but the chip is on board as you can see :   

[image]

another thing is that the oled display didn't poweron

 - OS: Win10
 - PlatformIO Version: last available
 - Firmware Version:  last available

thank you
73 de IW9GRL

iw9hfs, 2022-01-26:
I'm the friend :-) 73 IW9HFS Francesco.ciao

peterus, 2022-02-17:
Hi all,
sorry for the big delay.
Try to erase the board and reflash it again. Try to get the log output when you first startup the board to see if the board finding is successful.
73 de OE5BPA

iw9grl, 2022-02-23:
> Hi all, sorry for the big delay. Try to erase the board and reflash it again. Try to get the log output when you first startup the board to see if the board finding is successful. 73 de OE5BPA

Hi

thank you for the answer, we will try with your suggestion . In case of other fault I will copy/paste the log here !

73 de IW9GRL

closed 2022-05-07 by peterus