lora-aprs/LoRa_APRS_iGate #158: Internal OLED Display not functioning | Heltec Lora 32 V2

DL5SMB, 2022-02-19:
Hi everyone,

I have a brand new Heltec Lora ESP 32 V2 whose internal OLED display refuses to work after flashing.
The display itself should be okay, as it worked with the original  firmware it was shipped with.

Using the serial monitor  I can't see any error messages, 
except (after while):

**[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit**

The module connects to my wifi and sends its position, just no display. So I think this is the only problem.

Any idea where the problem might be here? I'm using FW 21.50.0 and i think also the newest Vusual Stuidion / Platformio version.

Sascha, DL5SMB

DL5SMB, 2022-02-19:
Okay, it seems as if the wrong board gets detected.
Sorry, should have noticed this earlier

> I      LoRa_APRS_iGate.cpp   47 : LoRa APRS iGate by OE5BPA (Peter Buchegger)
> I       LoRa_APRS_iGate.cpp   48 : Version: 21.50.0
> **I       LoRa_APRS_iGate.cpp   79 : Board TTGO_LORA32_V2 loaded.**`

DL5SMB, 2022-02-20:
And I am one step closer to the final solution.

After reflashing I noticed  other error messages following the first boot,
when the hardware is detected and set:

> Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> I       LoRa_APRS_iGate.cpp   47 : LoRa APRS iGate by OE5BPA (Peter Buchegger)
> I       LoRa_APRS_iGate.cpp   48 : Version: 21.50.0
> W lib\ConfigurationManagement\configuration.cpp   27 : Failed to read file, using default configuration.
> I lib\BoardFinder\BoardFinder.cpp   13 : looking for a board config.
> I lib\BoardFinder\BoardFinder.cpp   14 : searching for OLED...      
> **[E][esp32-hal-i2c.c:1434] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=0, scl=0**
> **W lib\BoardFinder\BoardFinder.cpp   73 : issue with wire
> D lib\BoardFinder\BoardFinder.cpp  123 : -1
> D lib\BoardFinder\BoardFinder.cpp  128 : power chip NOT found**
> I lib\BoardFinder\BoardFinder.cpp   26 : found a board config: TTGO_LORA32_V2
> I       LoRa_APRS_iGate.cpp   74 : will restart board now!
> I       LoRa_APRS_iGate.cpp   47 : LoRa APRS iGate by OE5BPA (Peter Buchegger)
> I       LoRa_APRS_iGate.cpp   48 : Version: 21.50.0
> I       LoRa_APRS_iGate.cpp   79 : Board TTGO_LORA32_V2 loaded. 

So after some googling I found a [post](https://github.com/cyberman54/ESP32-Paxcounter/issues/311) where a user adviced to connect GPIO21 (Vext power) to GND.

This is worked at least partially for me. 

Now instead of a TTGO_LORA32_V2 a Heltec WiFI LoRa  32v **_1_** gets detected. 

I do have a working OLED-display now, but as the pins for LoRa changed between [V1 and V2](https://heltec-automation-docs.readthedocs.io/en/latest/esp32/wifi_lora_32/hardware_update_log.html#v2) (which i have)  LoRa is not working yet.

I guess I could change the pins in the code, but it's probably more reasonable to find out where the real problem is and how to solve it. 

Where might this detection problem be? Is this somehow connected to pull-ups?

Or how are a V1, a V2 and a V2.1 distinguished?  I am afraid my coding skills are not good enough to understand how the detection works in the source code.

Also I think  it might really make sense to get an option to override detection with a configuration flag in the is-cfg.json.

peterus, 2022-02-21:
Thanks for your detailed explanation.

1) can you post a link with our exact board? I was always thinking it is the same as the one from Lilygo. I will buy one to have one for testing ;)

2) to fix your current issue you can force the board in the config file `is-cfg.json`:
add the `board` setting to the config:
```
	},
	"ntp_server": "pool.ntp.org",
        "board": "HELTEC_WIFI_LORA_32_V2"
}
```
be sure everything is in correct json format (don't forget the `,` !).

If this is not working, activate the ftp server, download the config via an ftp client (simple ftp, no sftp oder ftps!), change the board name and upload it again to the board.

DL5SMB, 2022-02-21:
Hi Peterus,

thanks for your answer.

[This](https://www.aliexpress.com/item/33018609928.html?spm=a2g0o.order_detail.0.0.18036368PfLIOj is the board I bought.

I'll check what happens if I select the board according to your instructions in the config-file.

closed 2022-05-07 by peterus