lora-aprs/LoRa_APRS_iGate #169: assertion "Invalid mbox" failed => tcpip.c

hb9eue, 2022-03-23:
Hallo zusammen

Wollte soeben auf die Version 22.11.2 updaten...

[DEBUG][TaskManager] call setup for RouterTask
[DEBUG][TaskManager] call setup for OTATask
[DEBUG][TaskManager] call setup for NTPTask
[DEBUG][TaskManager] call setup for AprsIsTask
[INFO][Main] setup done...
[INFO][RouterTask] [00:00:06] HB9EUE-L>APLG01:=4726.56NL00819.75E&LoRa iGATE & Digi, Info: github.com/lora-aprs/LoRa_APRS_iGate
assertion "Invalid mbox" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c", m
abort() was called at PC 0x400eb4f3 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x400891fc:0x3ffb1d60 0x40089479:0x3ffb1d80 0x400eb4f3:0x3ffb1da0 0x40134873:0x3ffb1dd0 0x40133fd9:0x3ffb1e00 0x4013418c:0x3ffb1e20 0x4013881a:0x0

Rebooting...

hb9eue, 2022-03-23:
My fault, should have diffed is-config.js and taken the changes to the new FS image.

peterus, 2022-03-23:
hi @hb9eue !
I will reopen the issue as this should not happen on a simple version update.
Can you confirm the missing setting for: syslog -> server?
I think if this field is missing it is throwing this assertion.

hb9eue, 2022-03-23:
Yes, the syslog config was missing and the network part with static IP addresses.

peterus, 2022-04-07:
fixed with #181

closed 2022-04-07 by peterus