lora-aprs/LoRa_APRS_iGate #188: No radio on my TTGO T-Beam V1 (433MHz SX1278)

m1geo, 2022-04-21:
Hey.

I have two TTGO T-Beam V1 (433MHz SX1278) boards, one configured as a tracker and another as an iGate. The tracker seems to work correctly. 

However, the iGate doesn't appear to have working radio. It doesn't receive the tracker even over just a few metres. I can see the tracker transmitting on 433.775 MHz with a spectrum analyser. I switched the hardware and can confirm that the LoRa radio works on both boards (and the broken behaviour follows software). The iGate code _never_ receives the tracker and its beacons in digi mode never produce RF on a known working board. I therefore suspect I have a setting wrong of the code has a bug? Using [a599eda](https://github.com/lora-aprs/LoRa_APRS_iGate/commit/a599edabe29adea1ed0b08aff77cee505f5a4d14). The splashscreen says 22.14.0.

Any help greatly appreciated.
George, M1GEO.

wolfpcgn, 2022-04-24:
I use a T-BEAM V1.0 (without display) with this software - version as LoRaAPRS igate. It seems to work fine, with an antenna on the roof I received a station 6 km away.

m1geo, 2022-04-24:
Strange. I didn't see any RF coming out when it said "BEACON" on either board. And when the tracker beaconed (with RF) there was nothing received over about 1 metre.

wolfpcgn, 2022-04-24:
Looks like you are trying to use a T-BEAM for 868 MHz on 433 MHz. If this is the case, then the wrong RFout of the SX-chip is used.

m1geo, 2022-04-24:
But I'm not. Trust me. I can tell the difference between 433 and 868 ;) The two boards are identical, and if you read my original post you'll see I confirmed both will transmit on 433 MHz using a spectrum analyser. The hardware is fine. Both boards transmit with the LoRa_APRS_Tracker code. Neither will transmit (or receive) with the aforementioned commit of LoRa_APRS_iGate.

peterus, 2022-05-07:
If there is a config issue, please re-download the config and update it again. you can also post your config here that we chan check it.

m1geo, 2022-05-18:
Still the same with a completely new download and config file.

daniel-hofer, 2022-06-07:
> Looks like you are trying to use a T-BEAM for 868 MHz on 433 MHz. If this is the case, then the wrong RFout of the SX-chip is used.
Sorry, litte off-topic:
Hi, I'm trying to enable my 868Mhz Devices. Is this generally possible? Got errors about unsopported frequency when I change to 868 in the config. If not, I stuck on meshtastic and have to buy new devices for aprs.

m1geo, 2022-06-10:
Nope, I’m using 433 MHz devices, as I said. I confirmed this with a spectrum analyser. The units also work fine with different firmware, so I am sure the hardware itself is working 100%.

 

George M1GEO.

Reply to: lora-aprs/LoRa_APRS_iGate ***@***.***>
Date: Tuesday, 7 June 2022 at 07:38
To: lora-aprs/LoRa_APRS_iGate ***@***.***>
Cc: "Dr. George Smart" ***@***.***>, Author ***@***.***>

BH6IDD, 2023-12-13:
The same situation applies. I'm using the TTGO LoRa32 V1 (433MHz SX1278) hardware, flashed with firmware version v23.31.1.

After flashing, it transmits its own beacon to the APRS-IS server. However, it is unable to receive APRS beacons sent by other handheld devices. When another handheld device transmits APRS beacons nearby on the same frequency, the iGate shows no response, and no data is being uploaded to the server.

RFRanger, 2024-01-15:
Same for me.  TTGO LoRa32 V1.  Everything seems to be working fine and the iGate is displayed on aprs.fi, but when I beacon to it from any other APRS transmitter, the iGate doesn't do anything and nothing is shown on aprs.fi.  Thanks for a great project.