lora-aprs/LoRa_APRS_iGate #197: Adding Retain-Flag, RSSI, SNR and FreqError to MQTT Message

Mane76, 2022-06-06:
Currently when a packet is received and sent to APRS-IS you can also configure to send the event to syslog. In syslog RSSI, SNR and FreqError is sent along with the message.
If MQTT is used only the event along with the message is sent without the retain-flag.

It would be great if
1) the retain flag can be set when sending to MQTT server, so that also new clients can receive the history
2) RSSI, SNR and FreqError will be added to be sent to MQTT server

Best regards
Manfred

Marcel-os, 2022-06-10:
Also it will be great if You could add RSSI of the WiFI signal to the MQTT packet.

73,
SP6MD

ploeffler, 2023-03-26:
...and maybe using homie-convention would ease-up things on the mqtt-side