lora-aprs/LoRa_APRS_iGate #200: [ERROR][RadiolibTask] readData failed, code -24

ok1mgj, 2022-06-08:
Hello,
i`m trying to run aprs igate on DollaTek LoRa32 V2.1 433 MHz ESP32, but i have one issue.
igate doesnt receive any data.
Beacon is  working fine
when i tryed another (digipeater only) sw, receiving was correctly working, so this is not HW issue.

into platformio.ini i have added section:
```
[env:ttgo-lora32-v21]
platform = espressif32
board = ttgo-lora32-v21
```
and filled is-config.json

next uploaded system and config.

Output on console is:
```...
...
...
[INFO][BeaconTask] [00:00:09] OK1MGJ-12>APLG01:=5030.22NL01326.61E&LoRa iGATE Test
[INFO][NTPTask] Current time: 06:07:54
[INFO][AprsIsTask] connecting to APRS-IS server: euro.aprs2.net on port: 14580
[INFO][AprsIsTask] Connected to APRS-IS server!
[ERROR][RadiolibTask] [06:08:43] readData failed, code -24
[INFO][NTPTask] Current time: 06:08:54
[INFO][NTPTask] Current time: 06:09:54
[ERROR][RadiolibTask] [06:10:17] readData failed, code -24
[ERROR][RadiolibTask] [06:10:35] readData failed, code -24
```

Propably theese errors are received packets wich wasn`t processed correctly.

any idea?

PS: Sorry me for my english, it`s not my native language.

peterus, 2022-06-09:
you do not need to change something in the platformio.ini! please leave it as it is and use the "default" target.

the firmware should look for the best board specifications for you.

do the following:
- delete the flash
- upload filesystem
- upload firmware and start the serial monitor directly after the upload (there is somewhere a button which will do this two steps in one).
- copy the first part until the first reboot and log it here.
- try it again.

ok1mgj, 2022-06-11:
im really sorry, just closing this issue.

Trouble wasnt neither lora module nor softare, but power source. Already has been trashed.

Many thanks for help.

closed 2022-06-11 by ok1mgj