lora-aprs/LoRa_APRS_iGate #201: NTP:Booting - no connection

dm6tt, 2022-06-16:
**Describe the bug**
NTP will stay in state booting, no Log message, no connection

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[INFO][Main] LoRa APRS iGate by OE5BPA (Peter Buchegger)
[INFO][Main] Version: 22.20.0
[INFO][Main] Board ETH_BOARD loaded.
[INFO][TaskManager] will setup all tasks...
[DEBUG][TaskManager] call setup for WifiTask
[INFO][WiFiEvent] WiFi Started
[DEBUG][WifiTask] Looking for AP: WLAN
[DEBUG][WifiTask] Looking for AP: xxxxxxx
[DEBUG][TaskManager] call setup for EthTask
[INFO][WiFiEvent] WiFi Started
[INFO][WiFiEvent] WiFi Started
[DEBUG][TaskManager] call setup for DisplayTask
[DEBUG][TaskManager] call setup for RadiolibTask
[DEBUG][TaskManager] call setup for RouterTask
[DEBUG][TaskManager] call setup for BeaconTask
[DEBUG][TaskManager] call setup for OTATask
[DEBUG][TaskManager] call setup for NTPTask
[DEBUG][TaskManager] call setup for FTPTask
[DEBUG][FTPTask] Adding user to FTP Server: ftp
[DEBUG][TaskManager] call setup for AprsIsTask
[INFO][WiFiEvent] ETH Connected
[INFO][WiFiEvent] ETH Connected
[INFO][Main] setup done...
[INFO][WiFiEvent] WiFi Connected
[INFO][WiFiEvent] WiFi Connected
[INFO][WiFiEvent] WiFi MAC: 34:94:54:5F:48:78
[INFO][WiFiEvent] IPv4: 192.168.3.xxx
[INFO][WiFiEvent] Gateway: 192.168.3.1
[INFO][WiFiEvent] DNS1: 192.168.3.1
[INFO][WiFiEvent] DNS2: 0.0.0.0
[INFO][WiFiEvent] WiFi MAC: 34:94:54:5F:48:78
[INFO][WiFiEvent] IPv4: 192.168.3.xxx
[INFO][WiFiEvent] Gateway: 192.168.3.1
[INFO][WiFiEvent] DNS1: 192.168.3.1
[INFO][WiFiEvent] DNS2: 0.0.0.0
[DEBUG][WifiTask] IP address: 192.168.3.xxx
[INFO][BeaconTask] [00:00:11] DM6TT-11>APLG01:=5013.98NL00726.46E&LoRa iGATE & Digi, Info: github.com/lora-aprs/LoRa_APRS_iGate

**Desktop (please complete the following information):**
 - OS: 
 - PlatformIO Version: 
 - Firmware Version: 

**Board name:**
Eth32 POE Board
- 

**Additional context**
The iGate is not connecting to NTP.
Tried with different NTP configurations and different networks.
Config Line:
"ntp_server": "ptbtime1.ptb.de"

It worked for a long time, no clue what happend.
I reflashed the board with the newest firmware.

dahuafschmied, 2022-06-24:
there is a problem with DNS / name resolution
syslog does not resolve and ntp to.

just enter the ip address of an NTP server instead of the name in the filed and it is working again....

peterus, 2022-07-05:
thx for the fix!

closed 2022-07-05 by peterus