lora-aprs/LoRa_APRS_iGate #208: TTGO T-Beam V1.1 Supported

erikdewildt, 2022-07-30:
Hi,

The docs mention support for the "TTGO T-Beam V1 (433MHz SX1278)". Is the 1.1 version of the board also supported? Those seem to be the only ones i can buy on AliExpress. [Link](https://nl.aliexpress.com/item/32967228739.html?spm=a2g0o.cart.0.0.2a2161d7mmdone&mp=1&gatewayAdapt=glo2nld)

Also i noticed a version of the  TTGO T-Beam V1.1 with a SX1268 chip. This one has a bit more power and lower current draw. Is this chip supported as well?

Thanks,
Regards,

PE1PUX, Erik

wolfpcgn, 2022-08-07:
imho the version 1.1 should be supported because the only difference seems to be the protection against wrong insertion of the battery.
I don't know about a SX1268-version, where can I buy it?

yooshky, 2022-10-12:
Does it work fine on the SX1268? How to do it?

peterus, 2022-10-17:
SX1268 is still not supported, but I have a board at home to test this out. please stay tune until I have the correct changes.

kratinoska, 2022-11-01:
I tried iGate and Tracker and t-beam with SX1268 does not work.
serial monitor on Tracker:
LoRa_APRS_Tracker.cpp  384 : Starting LoRa failed!

Ingineerix, 2023-08-16:
Looks like the v1.1 also uses the AXP2101 power controller, which doesn't initialize: "AXP192 init failed!"
Seems like the fix would be to migrate to the XpowersLib which can do the new chip and the older ones.

kratinoska, 2023-08-17:
My v1.1 SX1268 uses AXP192.

ab0oo, 2023-08-18:
There's some mis-labeling going on with the new T-Beams.  The silkscreen on the board says "TBeam -AXP2102-v1.2".  The big shiny sticker on the end of the board says Model: T-Beam-V1.1.  It also comes with a Neo-M8N GPS module.

kratinoska, 2023-08-18:
version of my board is T22_V1.1 20210222 (SX1268,, GPS NEO-M8N, IPEX )

Ingineerix, 2023-08-18:
Mine is like [ab0oo](https://github.com/ab0oo)'s above, and definitely has the AXP2101 PMIC, so incompatible.   [Picture Here](https://photos.app.goo.gl/QMX4mnwjuo5Xc1My5)