lora-aprs/LoRa_APRS_iGate #214: Using T-Beam V1.1: How to select the right board

dl5dla, 2022-09-05:
Hi,
I have compiled and installed the software on a T-Beam V1.1 without display. The monitoring shows:
"Board TTGO_LORA32_V1 loaded", but I would have expected a "TTGO_T_Beam_V1_0" instead.

The quick start guide mentions, that I have to select the right board in the status bar at the bottom, but the required "env:ttgo-t-beam-v1" is not listed as an option (and also not in the platformio.ini). So I am a little bit lost currently.
Thanks,
 Peter

dl5dla, 2022-09-06:
After stepping through the source code, I found that I have to add
`"board": "TTGO_T_Beam_V1_0"`
into the file is-cfg.json to enable the device. I also addeAfter stepping through the source code, I found that I have to add
`"board": "TTGO_T_Beam_V1_0"`
into the file is-cfg.json to enable the device. I also included
`[env:ttgo-t-beam-v1]
board = ttgo-t-beam
build_flags = -Werror -Wall`
into the file platformio.ini (not sure whether this is really needed. 

Please let me know there is a better way to solve it.

peterus, 2022-10-17:
do you have an oled attached to the board? if not, the firmware has issues to find the correct board sometimes. In this case it is totally okay to add the mentioned line to the json config.

you do not need to change something in the ini-file as the firmware will try to find the correct board configuration for you.

closed 2022-10-17 by peterus