lora-aprs/LoRa_APRS_iGate #229: Unable to select boards other than lora_board and lora_board_debug

FrankHovis, 2022-12-19:
Docs state that 
You can use one of the Lora32 boards without changes:

    Heltec WiFi LoRa 32 V1 (433MHz SX1278)
    Heltec WiFi LoRa 32 V2 (433MHz SX1278)
    TTGO LoRa32 V1 (433MHz SX1278)
    TTGO LoRa32 V2 (433MHz SX1278)
    TTGO LoRa32 V2.1 (433MHz SX1278)
    etc.

but I'm only seeing 
 Default
 env:lora_board
 env:lora_board_debug

Any clue as to how to get a different board?

peterus, 2022-12-19:
just use the default configuration in platformIO. the firmware will try to find out which board you are using. No need to select something.

closed 2022-12-19 by peterus

FrankHovis, 2022-12-25:
Unfortunately, it isn't as I'm using a separate ESP32 and Lora module. How can I specify a particular board?