lora-aprs/LoRa_APRS_iGate #230: add  heltec_wifi_lora_32_V3 support

octavia75, 2022-12-19:
hello can you add support for heltec_wifi_lora_32_V3

peterus, 2022-12-19:
can you try to upload the firmware and capture the output from the first startup when the firmware is trying to find out which board you are using. I have the feeling the board is already supported but not in the list.

frankwatervoort, 2023-02-14:
If you could let me know exactly what you want me to do, I can give it a try. Do you want me to build/compile for a Heltec Wifi Lora 32 V2 board and try and upload it to the V3 board? We tried that before. You get an Invalid head of packet (0x47) during upload. Thanks for your help in advance! I tried using [this article](https://community.platformio.org/t/heltec-esp32-lora-v3-board-support/30406) as well, but it results in all kinds of errors during the build process.

frankwatervoort, 2023-02-14:
So, I added this section below to the platformio.ini for the Lora_APRS_iGate, as seen in the article in my previous post, but the result is that the build process failed when selecting the helect_wifi_lora_32_v3 board in Platformio.

[env:heltec_wifi_lora_32_V3] 
platform = https://github.com/Baptou88/platform-espressif32.git
framework = arduino
board = heltec_wifi_lora_32_V3
platform_packages = framework-arduinoespressif32@https://github.com/Baptou88/arduino-esp32.git

Thanks for your help!