lora-aprs/LoRa_APRS_iGate #261: PoE Ethernet board: network via Eth not working

peterus, 2023-02-05:
On the PoE Ethernet board the current (23.04.0) version is not working. The Ethernet connection is not recognized, so the firmware is not going into the "connected" state and the data is not sent to the aprs-is servers.
It looks like it is getting an Ip because the router is recognizing the board and is assigning an Ip.

For me this looks like there was a change in the Arduino Core for the Ethernet part.

dahuafschmied, 2023-02-06:
and this is with PoE connected to Ethernet:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
[INFO][Main] LoRa APRS iGate by OE5BPA (Peter Buchegger)
[INFO][Main] Version: 23.04.0
[INFO][Main] Board ETH_BOARD loaded.
[INFO][Main] Will start watchdog now...
[INFO][TaskManager] will setup all tasks...
[DEBUG][TaskManager] call setup for WifiTask
[INFO][WiFiEvent] WiFi Started
[DEBUG][WifiTask] Looking for AP: IOT
[DEBUG][TaskManager] call setup for EthTask
[DEBUG][TaskManager] call setup for DisplayTask
[DEBUG][TaskManager] call setup for RadiolibTask
[DEBUG][TaskManager] call setup for RouterTask
[DEBUG][TaskManager] call setup for BeaconTask
[DEBUG][TaskManager] call setup for OTATask
[DEBUG][TaskManager] call setup for NTPTask
[DEBUG][TaskManager] call setup for AprsIsTask
[INFO][Main] setup done...
[INFO][WiFiEvent] WiFi Connected
[INFO][WiFiEvent] WiFi Connected
E (18878) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (18878) task_wdt:  - loopTask (CPU 1)
E (18878) task_wdt: Tasks currently running:
E (18878) task_wdt: CPU 0: IDLE
E (18878) task_wdt: CPU 1: IDLE
E (18878) task_wdt: Aborting.

abort() was called at PC 0x400fa7e0 on core 0

Backtrace: 0x40083c7d:0x3ffbeccc |<-CORRUPTED

ELF file SHA256: 9d7ef034838ad48c

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
[INFO][Main] LoRa APRS iGate by OE5BPA (Peter Buchegger)
[INFO][Main] Version: 23.04.0
[INFO][Main] Board ETH_BOARD loaded.
[INFO][Main] Will start watchdog now...
[INFO][TaskManager] will setup all tasks...
[DEBUG][TaskManager] call setup for WifiTask
[INFO][WiFiEvent] WiFi Started
[DEBUG][WifiTask] Looking for AP: IOT
[DEBUG][TaskManager] call setup for EthTask
[DEBUG][TaskManager] call setup for DisplayTask
[DEBUG][TaskManager] call setup for RadiolibTask
[DEBUG][TaskManager] call setup for RouterTask
```

peterus, 2023-02-06:
thanks for the input @dahuafschmied 
can you check this PR on your side? https://github.com/lora-aprs/LoRa_APRS_iGate/pull/262
this should fix the issue with the missing ethernet connection.

dahuafschmied, 2023-02-06:
sorry. still reboots

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
[INFO][Main] LoRa APRS iGate by OE5BPA (Peter Buchegger)
[INFO][Main] Version: 23.04.0
[INFO][BoardFinder] looking for a board config.
[INFO][BoardFinder] searching for OLED...
[  1682][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[DEBUG][BoardFinder] wire response: -1
[DEBUG][BoardFinder] power chip NOT found
[  2697][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[DEBUG][BoardFinder] wire response: -1
[DEBUG][BoardFinder] power chip NOT found
[INFO][BoardFinder] found a board config: ETH_BOARD
[INFO][Main] will restart board now!
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
[INFO][Main] LoRa APRS iGate by OE5BPA (Peter Buchegger)
[INFO][Main] Version: 23.04.0
[INFO][Main] Board ETH_BOARD loaded.
[INFO][Main] Will start watchdog now...
[INFO][TaskManager] will setup all tasks...
[DEBUG][TaskManager] call setup for WifiTask
[INFO][WiFiEvent] WiFi Started
[DEBUG][WifiTask] Looking for AP: IOT
[DEBUG][TaskManager] call setup for EthTask
[INFO][WiFiEvent] ETH Started
[INFO][WiFiEvent] ETH Started
[DEBUG][TaskManager] call setup for DisplayTask
[DEBUG][TaskManager] call setup for RadiolibTask
[DEBUG][TaskManager] call setup for RouterTask
[DEBUG][TaskManager] call setup for BeaconTask
[DEBUG][TaskManager] call setup for OTATask
[DEBUG][TaskManager] call setup for NTPTask
[DEBUG][TaskManager] call setup for AprsIsTask
E (21596) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (21596) task_wdt:  - loopTask (CPU 1)
E (21596) task_wdt: Tasks currently running:
E (21596) task_wdt: CPU 0: IDLE
E (21596) task_wdt: CPU 1: IDLE
E (21596) task_wdt: Aborting.

abort() was called at PC 0x400fa7a8 on core 0

Backtrace: 0x40083c7d:0x3ffbeccc |<-CORRUPTED

ELF file SHA256: f6bd3acf7fe39a5e

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
[INFO][Main] LoRa APRS iGate by OE5BPA (Peter Buchegger)
[INFO][Main] Version: 23.04.0
[INFO][Main] Board ETH_BOARD loaded.
[INFO][Main] Will start watchdog now...
[INFO][TaskManager] will setup all tasks...
[DEBUG][TaskManager] call setup for WifiTask
[INFO][WiFiEvent] WiFi Started
[DEBUG][WifiTask] Looking for AP: IOT
[DEBUG][TaskManager] call setup for EthTask
```

dahuafschmied, 2023-02-06:
setup takes to long for wdg.
`  //delay(5000);`

i removed the delay line 176 now it is up and running

peterus, 2023-02-06:
you are right! this is way too long! no idea why this is still inside 😂
will remove it 👍

PR #262 (2023-02-06): fix eth issue
fix #261
closed 2023-02-06 by peterus