lora-aprs/LoRa_APRS_iGate #276: Destination APLG01, path Type Error Data

blacksocket, 2023-03-12:
Hello,

I have compiled and uploaded the latest firmware and during the start I am getting the error "Destination APLG01, path Type Error Data" like on the belo image. Then it can connect to aprs.fi and the Lora Gateway is also visible on the map. However when launch the second device T-Beam v1 board with LORA APRS tracker i can't see any data received by the LORA APRS iGate.

**Expected behavior**
The LORA Tracker should be received by LORA iGate and should be visible in aprs.fi map

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - PlatformIO Version: 3.0.0
 - Firmware Version: v23.6.0

**Board name:**
TTGO T-Beam V1.1 ESP32 433 >> iGate
TTGO T-Beam V1.1 ESP32 433 >> Tracker

**Additional context**

I can't see the LED blinking on my tracker device like here:

https://youtu.be/7rNx44RMTtg?t=1985

[image]

blacksocket, 2023-03-13:
I have found root cause namely there were some connection issues to GPS module. That causing the tracker was not sending the signal. Now it works, so we can close the issue.

closed 2023-03-13 by blacksocket

videobuff, 2024-05-16:
Could you explain more in detail what kind of connection issues and what you did to correct them. Soldering ?
I have the same error but iwhen listening on the frequency by a Yeasu FT991A, i do not hear any beacon signals.