lora-aprs/LoRa_APRS_iGate #290: booting loop

Zoe1212Zoe, 2023-04-22:
i have the board ttgo LoRa32 v2.
if i set aprs_is
active to true,my board goes in a booting loop. 
when i set i to false, it boots but no sign of any beacon being sent.

my settings are:
{
	"callsign": "13RF1244-10",
	"network": {
		"DHCP": true,
		"static": {
			"ip": "192.168.0.100",
			"subnet": "255.255.255.0",
			"gateway": "192.168.0.1",
			"dns1": "192.168.0.1",
			"dns2": "192.168.0.2"
		},
		"hostname": {
			"overwrite": false,
			"name": "13RF1244-10"
		}
	},
	"wifi": {
		"active": true,
		"AP": [
			{
				"SSID": "CampPro-AF0329",
				"password": "19RF1244"
			}
		]
	},
	"beacon": {
		"message": "LoRa iGATE , Info: github.com/lora-aprs/LoRa_APRS_iGate",
		"position": {
			"latitude":51.54192,
			"longitude": 10.92011
		},
		"use_gps": false,
		"timeout": 15
	},
	"aprs_is": {
		"active":false,
		"passcode": "5521",
		"server": "euro.aprs2.net",
		"port": 14580,
		"filter": ""
		

	},
	"digi": {
		"active": true,
		"beacon": true
	},
	"lora": {
		"frequency_rx": 433775000,
		"gain_rx": 20,
		"frequency_tx": 433775000,
		"power": 10,
		"spreading_factor": 12,
		"signal_bandwidth": 125000,
		"coding_rate4": 5,
		"tx_enable": true
	},
	"display": {
		"always_on": true,
		"timeout": 10,
		"overwrite_pin": 0,
		"turn180":true
	},
	"ftp": {
		"active": false,
		"user": [
			{
				"name": "ftp",
				"password": "ftp"
			}
		]
	},
	"mqtt": {
		"active": false,
		"server": "",
		"port": 1883,
		"name": "",
		"password": "",
		"topic": "LoraAPRS/Data",
		"will_active": false,
		"will_topic": "LoraAPRS/State",
		"will_message": "offline",
		"birth_message": "online"
	},
	"syslog": {
		"active": false,
		"server": "",
		"port": 514
	},
	"ntp_server": "pool.ntp.org"
}

Zoe1212Zoe, 2023-04-22:
by tying to sent beacon its says: path:, type:error

peterus, 2023-04-22:
you can ignore the path error, this is known and is just a display issue.

can you open the serial monitor and copy and paste some lines into this issue?

closed 2023-04-22 by peterus

peterus, 2023-04-23:
@Zoe1212Zoe you dont need to provide me more details! I already found the bug 😉
update your code and flash your board again.

PR #291 (2023-04-22): Fix access to queue
close #290