lora-aprs/LoRa_APRS_iGate #296: Reboot Loop when enabling TX functionality

Mane76, 2023-05-18:
**Describe the bug**
When enabling TX functionality of iGate the code compiles, but the esp32 is in an reboot loop. By changing the line lora_tx_enable to false everything works fine as before

**To Reproduce**
Steps to reproduce the behavior:
1. Download code status 18.05.2023 from master branch
2. adjust config file, wifi active, aprsis active, digi false, lora tx enable, ftp false, mqtt false, syslog true
3. compile, upload to device, upload file system image 

**Expected behavior**
start as usual

**Screenshots**
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[INFO][Main] LoRa APRS iGate by OE5BPA (Peter Buchegger)
[INFO][Main] Version: 23.06.0
[INFO][Main] Board TTGO_LORA32_V2 loaded.
[INFO][Main] Will start watchdog now...
[INFO][TaskManager] will setup all tasks...
[DEBUG][TaskManager] call setup for WifiTask
[INFO][WiFiEvent] WiFi Started
[DEBUG][WifiTask] Looking for AP: xxxx
[DEBUG][TaskManager] call setup for DisplayTask
[DEBUG][TaskManager] call setup for RadiolibTask
[DEBUG][TaskManager] call setup for RouterTask
[DEBUG][TaskManager] call setup for BeaconTask
[DEBUG][TaskManager] call setup for OTATask
[DEBUG][TaskManager] call setup for NTPTask
[DEBUG][TaskManager] call setup for AprsIsTask
[INFO][Main] setup done...
E (22408) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (22408) task_wdt:  - loopTask (CPU 1)
E (22408) task_wdt: Tasks currently running:
E (22408) task_wdt: CPU 0: IDLE
E (22408) task_wdt: CPU 1: IDLE
E (22408) task_wdt: Aborting.

abort() was called at PC 0x400fb130 on core 0

Backtrace: 0x40083c1d:0x3ffbeccc |<-CORRUPTED

ELF file SHA256: d0eeed056324df7f

Rebooting...

**Desktop (please complete the following information):**
 - OS: Linux x64
 - PlatformIO Version: 1.77.0
 - Firmware Version: 

**Board name:**
TTGO LORA32

**Additional context**
Add any other context about the problem here.

peterus, 2023-05-20:
looks like something is taking to long and the wathdog is restarting the esp.

can you post your exact configuration? please remove your personal data (callsign, wifi settings, aprs-is passcode)

Mane76, 2023-05-20:
Here we are... 
when I change the "tx_enable" in section "lora" to "false" everything is working fine. 
The frequency 433.781 is the correction of the offset of "my" module to the "real" RX/TX frequency.
As I see in the preview the formatting gets lost.

{
	"callsign": "xxx",
	"network": {
		"DHCP": true,
		"static": {
			"ip": "192.168.0.100",
			"subnet": "255.255.255.0",
			"gateway": "192.168.0.1",
			"dns1": "192.168.0.1",
			"dns2": "192.168.0.2"
		},
		"hostname": {
			"overwrite": false,
			"name": "xxx"
		}
	},
	"wifi": {
		"active": true,
		"AP": [
			{
				"SSID": "xxx",
				"password": "xxx"
			}
		]
	},
	"beacon": {
		"message": "LoRa iGate v23.17.1m",
		"position": {
			"latitude": xxx,
			"longitude": xxx
		},
		"use_gps": false,
		"timeout": 15
	},
	"aprs_is": {
		"active": true,
		"passcode": "xxx",
		"server": "germany.aprs2.net",
		"port": 14580,
		"filter": "r/xxx/xxx/50 -t/tpo"
	},
	"digi": {
		"active": false,
		"beacon": false
	},
	"lora": {
		"frequency_rx": 433781000,
		"gain_rx": 0,
		"frequency_tx": 433781000,
		"power": 20,
		"spreading_factor": 12,
		"signal_bandwidth": 125000,
		"coding_rate4": 5,
		"tx_enable": true
	},
	"display": {
		"always_on": false,
		"timeout": 10,
		"overwrite_pin": 0,
		"turn180": true
	},
	"ftp": {
		"active": false,
		"user": [
			{
				"name": "ftp",
				"password": "ftp"
			}
		]
	},
	"mqtt": {
		"active": false,
		"server": "",
		"port": 1883,
		"name": "",
		"password": "",
		"topic": "LoraAPRS/Data",
		"will_active": false,
		"will_topic": "LoraAPRS/State",
		"will_message": "offline",
		"birth_message": "online"
	},
	"syslog": {
		"active": true,
		"server": "xxx",
		"port": 514
	},
	"ntp_server": "pool.ntp.org"
}

ploeffler, 2023-05-20:
for me the LoRa-TX is working like a charm!

pinkolin, 2023-05-24:
I have similar issue maybe with same board?:

`[INFO][WiFiEvent] WiFi IPv4: 192.168.88.31
[INFO][WiFiEvent] WiFi Gateway: 192.168.88.1
[INFO][WiFiEvent] WiFi DNS1: 8.8.8.8
[INFO][WiFiEvent] WiFi DNS2: 1.1.1.1
[INFO][WiFiEvent] WiFi Hostname: OK1PNK-10
[DEBUG][WifiTask] WiFi IP address: 192.168.88.31

assert failed: heap_caps_free heap_caps.c:360 (heap != NULL && "free() target pointer is outside heap areas")

Backtrace: 0x40083d95:0x3ffb1f90 0x4008e151:0x3ffb1fb0 0x40093ead:0x3ffb1fd0 0x40084282:0x3ffb2100 0x40093edd:0x3ffb2120 0x40166655:0x3ffb2140 0x400d7cf9:0x3ffb2160 0x400d8463:0x3ffb2180 0x40182062:0x3ffb2230 0x400d3302:0x3ffb2250 0x400f255d:0x3ffb2290

ELF file SHA256: a003e5a76581958a

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (314) psram: PSRAM ID read error: 0xffffffff
[INFO][Main] LoRa APRS iGate by OE5BPA (Peter Buchegger)
[INFO][Main] Version: 23.06.0
[INFO][Main] Board TTGO_LORA32_V2 loaded.
[INFO][Main] Will start watchdog now...
[INFO][TaskManager] will setup all tasks...
[DEBUG][TaskManager] call setup for WifiTask
[INFO][WiFiEvent] WiFi Started`

ohanssen, 2023-06-02:
I have the same issue with a LoRa32 board.

peterus, 2023-06-04:
issue was reproduced, fix is in: https://github.com/lora-aprs/LoRa_APRS_iGate/pull/303

peterus, 2023-06-04:
@ohanssen  @pinkolin @Mane76 please report back if TX is now working for you! will create based on the feedback a new release. Thank you very much!

Mane76, 2023-06-04:
reboot loop is solved **BUT**

when "tx_enable": true **no** RX of tracker is possible
when "tx_enable": false a RX of tracker **is** possible

when "tx_enable": true everything which passes the filter set in section aprs-is is TXd

'73, Mane

peterus, 2023-06-05:
@Mane76 I just tried to reproduce your issue, but couldn't. can you please retry again (delete flash first, than upload config and than firmware upload with monitoring)?
thanks!

peterus, 2023-06-05:
@Mane76 are you using different frequencies? possible fix would be: https://github.com/lora-aprs/LoRa_APRS_iGate/pull/304

peterus, 2023-06-05:
@Mane76 good found! have not seen this difference! Will need to check this, but i think most probably this is the issue 👍 will check tomorrow 👍

ohanssen, 2023-06-05:
I see the same here. The reboot problem is fixed but the RX doesnt work if TX is enabled. Digi section "active" and "beacon" are set to false.

peterus, 2023-06-08:
i refactored the RadioLib Task, added the newest version of RadioLib and now it is working as expected (on my side at least). please try the latest master again and give feedback! thanks!

changes: https://github.com/lora-aprs/LoRa_APRS_iGate/pull/305

ohanssen, 2023-06-08:
Still the same on my side: If TX is enabled, RX seems to be disabled.

peterus, 2023-06-09:
@ohanssen did you follow this steps: https://github.com/lora-aprs/LoRa_APRS_iGate/issues/296#issuecomment-1577284813
the configuration file changed! best is to create a completely new one.

peterus, 2023-06-09:
@beppelpo can you check if RX is still possible when TX is active? thanks!

beppelpo, 2023-06-09:
> @beppelpo can you check if RX is still possible when TX is active? thanks!

Unfortunately there is no LoRa station in the area, I was hoping to spread this technique among local OMs. But I have another identical board with the TinyGS console. Then I try to install this iGate and see if the two devices listen to each other. I'll keep you posted on developments. Good evening.
Beppe i3lpo

peterus, 2023-06-09:
Phu... slowly I am giving up on this topic as on my side everything is working correctly and I have no idea anymore.

So...
1. please send me some pictures of the hardware which you are using.
2. I need a a config file, and the log output.
3. Further please make sure that you are using the latest mainline version - in the next minute version 23.23.99 is published.

beppelpo, 2023-06-09:
Well, I've installed an identical copy of iGate on another identical LILYGO card. I only changed id on the second one (i3lpo-9) . Both receive and transmit (tx true), in the video made on -9 then you see the msg received from -10. Only question that "Type: error.data "....
Beppe i3lpo

Uploading video5771553211001343426.mp4…

peterus, 2023-06-09:
@beppelpo:
`Type: error` can be ignored, this „bug“ is well known and just on the display/logging.

can you post the logs from the serial monitor too? to make sure? can you activate the beacon mode (`send_on_hf: true`) on one board just to be sure? thanks!

Mane76, 2023-06-09:
Just tested, same behaviour. See pic from device attached. 
Can you provide a email adress for debug log ? A lot of pages...
[image]

peterus, 2023-06-09:
@Mane76 oe5bpa@gmail.com

beppelpo, 2023-06-09:
send_on_hf: true in both tabs, I'll see if I can send the log, I'm not very practical!

peterus, 2023-06-09:
@Mane76 I think your issue is now at this point something different than the original issue described here.

can you comment all lines which are responsible for the watchdog?
like in `LoRa_APRS_iGate.cpp` line 94 - 101, 154, 163, 184.

I think your esp needs more time to connect to your wifi and than the watchdog is kicking in.

Mane76, 2023-06-10:
commented watchdog lines, unfortunately no effect on RX problem

beppelpo, 2023-06-11:
[image]

I don't know if this img is enough, I can't activate the output to file. I'm sorry but I'm really a sucker with VSCode. Beppe i3lpo

beppelpo, 2023-06-11:
sucker = duffer