lora-aprs/LoRa_APRS_iGate #298: TTGO Lora32 V2.1 Does not boot image

DL4ZAQ, 2023-05-19:
**Describe the bug**

TTGO Lora32 V2.1 Does not boot image after upload via Upload File System image in PlatformIO.
Build ok in PlatformIO
Upload ok in PlatformIO
Board stays with the meshtastic firmware on it after it rebooted.
Tried it many times with always the same result.

**Desktop (please complete the following information):**
 - OS: Win 10
 - PlatformIO Version: Newest
 - Firmware Version: Newest

*  Executing task: C:\Users\cpnef\.platformio\penv\Scripts\platformio.exe run --target uploadfs --environment lora_board

Processing lora_board (board: ttgo-lora32-v2; platform: espressif32 @ 6.1.0; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://[docs.platformio.org/page/boards/espressif32/ttgo-lora32-v2.html](https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v2.html)
PLATFORM: Espressif 32 (6.1.0) > TTGO LoRa32-OLED V2
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
- framework-arduinoespressif32 @ 3.20007.0 (2.0.7)
- tool-esptoolpy @ 1.40500.0 (4.5.0)
- tool-mkfatfs @ 2.0.1
- tool-mklittlefs @ 1.203.210628 (2.3)
- tool-mkspiffs @ 2.230.0 (2.30)
- toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://[bit.ly/configure-pio-ldf](https://bit.ly/configure-pio-ldf)
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 50 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.21.2
|-- AXP202X_Library @ 1.1.3
|-- APRS-Decoder-Lib @ 0.0.6
|-- esp-logger @ 1.0.0
|-- ESP-FTP-Server-Lib @ 0.11.0
|-- PubSubClient @ 2.8.0
|-- TinyGPSPlus @ 1.0.3
|-- OneButton @ 1.5.0
|-- RadioLib @ 5.7.0
|-- APRS-IS
|-- WiFi @ 2.0.0
|-- BoardFinder
|-- Ethernet @ 2.0.0
|-- SPI @ 2.0.0
|-- Wire @ 2.0.0
|-- System
|-- Display
|-- ConfigurationManagement
|-- PowerManagement
|-- FS @ 2.0.0
|-- NTPClient
|-- ArduinoOTA @ 2.0.0
|-- Update @ 2.0.0
|-- TimeLib
|-- SPIFFS @ 2.0.0
Building in release mode
Building FS image from 'data' directory to .pio\build\lora_board\spiffs.bin
/is-cfg.json
Looking for upload port...
Auto-detected: COM7
Uploading .pio\build\lora_board\spiffs.bin
esptool.py v4.5
Serial port COM7
Connecting....
Chip is ESP32-PICO-D4 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: d4:d4:da:9d:84:70
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00290000 to 0x003effff..
Compressed [1441792](tel:1441792) bytes to 3639...
Writing at 0x00290000.. (100 %)
Wrote 1441792 bytes (3639 compressed) at 0x00290000 in 5.9 seconds (effective 1960.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
================================================================================== [SUCCESS] Took 15.40 seconds ==================================================================================
Environment    Status    Duration
-------------  --------  ------------
lora_board     SUCCESS   00:00:15.400
================================================================================== 1 succeeded in 00:00:15.400 ==================================================================================
 *  Terminal will be reused by tasks, press any key to close it.

**Additional context*
Followed your great installation guideline

DL4ZAQ, 2023-05-19:
BTW I changed the board and tried also V2.1, V1.0 as well as T-Beam (yes I am really lost at this point) without success. To me it seems that there is no data flashed to the Device… no matter what the terminal is telling me. At least it appears, that the currently installed meshtastic software is still there.
I also tried to do a flash erase - then the screen of the TTOG Device is black but it does not change it´s behavior after uploading the file system. I have no clue..

peterus, 2023-05-20:
Hi Peter!

To flash the board, you need to execute 2 steps (one time to upload the firmware, the second time is to upload the configuration).

You can find more here: https://www.lora-aprs.info/docs/LoRa_APRS_iGate/quick-start-guide/
or on some youtube videos (Manuel Lausmann, etc)

DL4ZAQ, 2023-05-20:
Hallo Peter,der Code kompiliert leider nicht sondern wird mit Fehlermeldung beendet:GrussPeter .   *  Executing task: C:\Users\cpnef\.platformio\penv\Scripts\platformio.exe run --environment lora_board  Processing lora_board (board: ttgo-lora32-v21; platform: espressif32 @ 6.1.0; framework: arduino)--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` optionCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v21.htmlPLATFORM: Espressif 32 (6.1.0) > TTGO LoRa32-OLED v2.1.6HARDWARE: ESP32 240MHz, 320KB RAM, 4MB FlashDEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)  PACKAGES:  - framework-arduinoespressif32 @ 3.20007.0 (2.0.7) - tool-esptoolpy @ 1.40500.0 (4.5.0) - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldfLDF Modes: Finder ~ deep+, Compatibility ~ softFound 50 compatible librariesScanning dependencies...Dependency Graph|-- ArduinoJson @ 6.21.2|-- AXP202X_Library @ 1.1.3|-- APRS-Decoder-Lib @ 0.0.6|-- esp-logger @ 1.0.0|-- ESP-FTP-Server-Lib @ 0.11.0|-- PubSubClient @ 2.8.0|-- TinyGPSPlus @ 1.0.3|-- OneButton @ 1.5.0|-- RadioLib @ 5.7.0|-- APRS-IS|-- WiFi @ 2.0.0|-- BoardFinder|-- Ethernet @ 2.0.0|-- SPI @ 2.0.0|-- Wire @ 2.0.0|-- System|-- Display|-- ConfigurationManagement|-- PowerManagement|-- FS @ 2.0.0|-- NTPClient|-- ArduinoOTA @ 2.0.0|-- Update @ 2.0.0|-- TimeLib|-- SPIFFS @ 2.0.0Building in release modeCompiling .pio\build\lora_board\src\LoRa_APRS_iGate.cpp.oCompiling .pio\build\lora_board\src\TaskAprsIs.cpp.oCompiling .pio\build\lora_board\src\TaskBeacon.cpp.oCompiling .pio\build\lora_board\src\TaskDisplay.cpp.oIn file included from C:/Users/cpnef/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:79,                 from C:/Users/cpnef/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,                 from .pio/libdeps/lora_board/esp-logger/src/logger.h:4,                 from src/TaskAprsIs.cpp:1:C:/Users/cpnef/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory *********************************************************************** Looking for pins_arduino.h dependency? Check our library registry!** CLI  > platformio lib search "header:pins_arduino.h"* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h***********************************************************************  #include "pins_arduino.h"          ^~~~~~~~~~~~~~~~compilation terminated.*** [.pio\build\lora_board\src\TaskAprsIs.cpp.o] Error 1In file included from C:/Users/cpnef/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:79,                 from C:/Users/cpnef/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,                 from .pio/libdeps/lora_board/esp-logger/src/logger.h:4,                 from src/TaskDisplay.cpp:1:C:/Users/cpnef/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory *********************************************************************** Looking for pins_arduino.h dependency? Check our library registry!** CLI  > platformio lib search "header:pins_arduino.h"* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h***********************************************************************  #include "pins_arduino.h"          ^~~~~~~~~~~~~~~~compilation terminated.In file included from C:/Users/cpnef/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:79,                 from C:/Users/cpnef/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,                 from .pio/libdeps/lora_board/esp-logger/src/logger.h:4,                 from src/TaskBeacon.cpp:1:C:/Users/cpnef/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory *********************************************************************** Looking for pins_arduino.h dependency? Check our library registry!** CLI  > platformio lib search "header:pins_arduino.h"* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h***********************************************************************  #include "pins_arduino.h"          ^~~~~~~~~~~~~~~~compilation terminated.*** [.pio\build\lora_board\src\TaskDisplay.cpp.o] Error 1*** [.pio\build\lora_board\src\TaskBeacon.cpp.o] Error 1In file included from C:/Users/cpnef/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:79,                 from C:/Users/cpnef/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,                 from .pio/libdeps/lora_board/APRS-Decoder-Lib/src/APRSMessage.h:4,                 from .pio/libdeps/lora_board/APRS-Decoder-Lib/src/APRS-Decoder.h:4,                 from lib/APRS-IS/APRS-IS.h:5,                 from src/LoRa_APRS_iGate.cpp:4:C:/Users/cpnef/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or dir [... cut ...] 9799 7410Email: ***@***.*** Von: Peter BucheggerGesendet: Samstag, 20. Mai 2023 07:38An: lora-aprs/LoRa_APRS_iGateCc: DL4ZAQ; AuthorBetreff: Re: [lora-aprs/LoRa_APRS_iGate] TTGO Lora32 V2.1 Does not boot image (Issue #298) Hi Peter!To flash the board, you need to execute 2 steps (one time to upload the firmware, the second time is to upload the configuration).You can find more here: https://www.lora-aprs.info/docs/LoRa_APRS_iGate/quick-start-guide/or on some youtube videos (Manuel Lausmann, etc)

DL4ZAQ, 2023-05-21:
Hallo Peter,nach langem Hin und Her bin ich einen Schritt weiter.Die Software kompiliert jetztAber: Der Upload funktioniert nicht… Im Dev. Manager wird das Board unter Port17 angezeigt. Fehler: Configuring upload protocol...AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpaCURRENT: upload_protocol = esptoolLooking for upload port...Using manually specified: COM17Uploading .pio/build/lora_board/firmware.binesptool.py v4.5Serial port COM17 A fatal error occurred: Could not open COM17, the port doesn't exist*** [upload] Error 2=============================================================================== [FAILED] Took 262.72 seconds =============================================================================== Environment    Status    Duration-------------  --------  ------------lora_board     FAILED    00:04:22.719=========================================================================== 1 failed, 0 succeeded in 00:04:22.719 =========================================================================== Ich bin mal wieder ratlos In der platformio.ini habe ich  Upload_port = COM17 Eingetragen… Unter Auto Detect funktioniert es auch nicht Mit Meshtastic kann ich das Board über den Webflasher problemlos flashen – es wird erkannt und keine Probleme Gibt es eine Möglichkeit, die kompilierte Software in einen File zu schreiben und mit einer anderen Methode das Board zu  flashen? GrussPeterDL4ZAQ  Von: Peter BucheggerGesendet: Samstag, 20. Mai 2023 07:38An: lora-aprs/LoRa_APRS_iGateCc: DL4ZAQ; AuthorBetreff: Re: [lora-aprs/LoRa_APRS_iGate] TTGO Lora32 V2.1 Does not boot image (Issue #298) Hi Peter!To flash the board, you need to execute 2 steps (one time to upload the firmware, the second time is to upload the configuration).You can find more here: https://www.lora-aprs.info/docs/LoRa_APRS_iGate/quick-start-guide/or on some youtube videos (Manuel Lausmann, etc)

DL4ZAQ, 2023-05-22:
Hallo Peter, vielen Dank für den Hinweise.
Zwischenzeitlich kompiliert die Software.
Leider kann ich die Software dann nicht auf das Board übertragen. PlatformIO erkennt den Upload Port nicht und der Vorgang wird abgebrochen. Im Win Dev. Manager ist das Board erkannt. Den COM Port habe ich dann auch in der PlatformIO.ini als Upload port "fest" verdrahtet... ohne Erfolg. 

Unter Linux habe ich das ganze dann auch noch mal aufgesetzt aber da kompiliert der Code schon gar nicht... Ich bekomme immer Fehlermeldungen und kaum habe ich eine Hürde genommen, kommt die nächste.

Das frustriert...
Ich habe das Board jetzt als GS-Groundstation geflashed - hat auf Anhieb funktioniert.
Als Meshtastic Node: auf Anhieb funktioniert.

Irgendwie habe ich jetzt nach vielen Stunden die Lust verloren.... und gebe auf. 
Warum muss das alles so kompliziert sein?

Danke und Gruss
Peter