lora-aprs/LoRa_APRS_iGate #326: Reboot loop no matter what

wyan, 2023-08-30:
**Describe the bug**
The Lora32 shows the boot screen but gets stuck in a boot loop.

**To Reproduce**
Steps to reproduce the behavior:
1. Check out commit `8c2c217acca4f951fc3583eed54a9c47e35e3e27` (tag `v23.31.1`, currently `HEAD` on master branch)
2. adjust config file, wifi active, aprsis active, everything else tried in many many combinations, all of them resulting in the same boot loop
3. compile, upload to device, upload file system image

**Expected behavior**
The board boots up correctly

**Screenshots**
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[INFO][Main] LoRa APRS iGate by OE5BPA (Peter Buchegger)
[INFO][Main] Version: 23.31.01
[INFO][Main] Board TTGO_LORA32_V2 loaded.
[INFO][Main] Will start watchdog now...
[INFO][TaskManager] will setup all tasks...
[DEBUG][TaskManager] call setup for WifiTask
[INFO][WiFiEvent] WiFi Started
[DEBUG][WifiTask] Looking for AP: finitud labs
[DEBUG][TaskManager] call setup for DisplayTask
[DEBUG][TaskManager] call setup for RadiolibTask
[DEBUG][RadiolibTask] [00:00:00] using SX1278
[DEBUG][TaskManager] call setup for RouterTask
[DEBUG][TaskManager] call setup for BeaconTask
[DEBUG][TaskManager] call setup for OTATask
[DEBUG][TaskManager] call setup for NTPTask
[DEBUG][TaskManager] call setup for FTPTask
[DEBUG][FTPTask] Adding user to FTP Server: admin
[DEBUG][TaskManager] call setup for AprsIsTask
[INFO][Main] setup done...
[INFO][WiFiEvent] WiFi Connected
E (22434) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (22434) task_wdt:  - loopTask (CPU 1)
E (22434) task_wdt: Tasks currently running:
E (22434) task_wdt: CPU 0: IDLE
E (22434) task_wdt: CPU 1: IDLE
E (22434) task_wdt: Aborting.

abort() was called at PC 0x400fef50 on core 0

Backtrace: 0x40083cfd:0x3ffbecdc |<-CORRUPTED

ELF file SHA256: 46b05c5ad1378826

Rebooting...
```

**Desktop (please complete the following information):**
 - OS: Ubuntu 23.04 on x64
 - PlatformIO Version: PlatformIO Core, version 6.1.10
 - Firmware Version: latest as of today

**Board name:**
Lilygo TTGO Lora32 T3_V1.6.2 (20210420)

**Additional context**
I've tried switching on and off the following: FTP, TX, send_on_hf, APRS-IS, all of them lead to the same boot loop.

wyan, 2023-08-30:
The config file used (private bits redacted out):
```
{
	"callsign": "EA4FYQ-10",
	"network": {
		"DHCP": true,
		"static": {
			"ip": "192.168.0.100",
			"subnet": "255.255.255.0",
			"gateway": "192.168.0.1",
			"dns1": "192.168.0.1",
			"dns2": "192.168.0.2"
		},
		"hostname": {
			"overwrite": false,
			"name": "EA4FYQ-10"
		}
	},
	"wifi": {
		"active": true,
		"AP": [
			{
				"SSID": "xxx xxx",
				"password": "xxx"
			}
		]
	},
	"beacon": {
		"message": "LoRa-iGate",
		"position": {
			"latitude": xxx
			"longitude": xxx
		},
		"use_gps": false,
		"timeout": 15,
		"send_on_hf": true
	},
	"aprs_is": {
		"active": true,
		"passcode": "xxx",
		"server": "euro.aprs2.net",
		"port": 14580,
		"filter": ""
	},
	"digi": {
		"active": false
	},
	"lora": {
		"frequency_rx": 433775000,
		"gain_rx": 0,
		"frequency_tx": 433775000,
		"power": 20,
		"spreading_factor": 12,
		"signal_bandwidth": 125000,
		"coding_rate4": 5,
		"tx_enable": false
	},
	"display": {
		"always_on": true,
		"timeout": 10,
		"overwrite_pin": 0,
		"turn180": true
	},
	"ftp": {
		"active": true,
		"user": [
			{
				"name": "xxx",
				"password": "xxx"
			}
		]
	},
	"mqtt": {
		"active": false,
		"server": "",
		"port": 1883,
		"name": "",
		"password": "",
		"topic": "LoraAPRS/Data",
		"will_active": false,
		"will_topic": "LoraAPRS/State",
		"will_message": "offline",
		"birth_message": "online"
	},
	"syslog": {
		"active": false,
		"server": "",
		"port": 514
	},
	"ntp_server": "pool.ntp.org"
}
```

noiasca, 2023-12-25:
Did you miss to uploade the config file? That's a separate task after flashing the program.
Go to the PlatformIO, PROJECT TASKS/lora_board_debug/Platform/Upload Filesystem Image