lora-aprs/LoRa_APRS_iGate #331: Heltec v3 board stuck in boot loop when connected to an USB power supply

dm5tt, 2023-09-19:
**Describe the bug**
When flashing the latest ARPS iGate firmware to a Heltec v3 AND connecting it to an USB power supply only the boot logo shows up and then reboots after a few seconds.

This NOT happens when connected to a PC.

**To Reproduce**

1. Flash latest 'master' firmware
2. Connect device to a USB power supply
3. Have a look at the screen and wait for reboot
 
**Expected behavior**
Boot up normally

**Screenshots**
If applicable, add screenshots to help explain your problem.

peterus, 2023-09-19:
did you check the power supply if it is producing the 5v correctly? can you try different power supplies? if it is working when connected to pc it will get very hard to debug.

dm5tt, 2023-09-19:
I checked with three power supplies and several cables. The Meshtastic firmware on that device seems not to cause any problems.

peterus, 2023-09-19:
thanks for your response! will need to check on my end.

dm5tt, 2023-09-19:
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/issues/159   wild guess, might be related

dm5tt, 2023-09-28:
I was able to narrow the problem down: The Heltec v3 is having an incredibly bad Wifi antenna. It was looping during the boot process as it wasn't able to get Wifi on my balcony.

closed 2023-09-28 by dm5tt