lora-aprs/LoRa_APRS_iGate #335: Liligo APX2101

f3boli, 2023-10-03:
Hello everyone. What could be the reason if my igate based on lilygo TBeam-APX2101-V1.2 is deaf for pure RX operation? It doesn't receive anything. I tried it with two different codes. First from Peter and then from Richonguzman. The tracker works on the same basis. But as an IGate it receives absolutely nothing.

peterus, 2023-10-04:
the board is still not completely supported by my firmware. It is already in preparation 😉

vk4ypb, 2023-11-11:
Turns out that my boards are these too. So would like this as well. The work you have done for this for the tracker is exceptional and works great. Hope you can get this up soon.
Peter.

Geissis, 2024-01-02:
Happy NEW YEAR to all
I am having the same board - when sending a package from tracker it shows the following error in Serial Monitor:
"[DEBUG][RadiolibTask] [10:08:34] Unknown packet '�HB9F' with RSSI -164dBm, SNR 0.00dB and FreqErr -0.000000Hz
[ERROR][RadiolibTask] [10:08:34] readData failed, code -24"
Maybe that helps to find the issue.
And many thanks to Peter for the great work!

RFRanger, 2024-01-15:
Ah, already commented on another thread (hadn't seen this one yet), but I have the LilyGO TTGO v1.6, and it's the same.  Thanks again for all of your work on this - it's very cool to see how far these LoRa boards will go! de ND4L