lora-aprs/LoRa_APRS_iGate #338: Welcome to LoRa_APRS_iGate Discussions!

woyteg, 2023-11-19:
### Discussed in https://github.com/lora-aprs/LoRa_APRS_iGate/discussions/101

<div type='discussions-op-text'>

<sup>Originally posted by **peterus** July 15, 2021</sup>

## 👋 Welcome!
  We’re using Discussions as a place to connect with other members of our community. We hope that you:
  * Ask questions you’re wondering about.
  * Share ideas.
  * Engage with other community members.
  * Welcome others and are open-minded. Remember that this is a community we
  build together 💪.

  To get started, comment below with an introduction of yourself, your hardware and antenna setup.

</div>

peterus, 2023-11-25:
next time please post the log directly in the ticket, attaching via a link a log is not nice (worms, spam, viruses, etc.)

you where to fast in installing platformio! remove your platformio installation and install it again AND take your time - platformio is doing a lot of stuff in the brackground which need to be finished bevor doing something else.
this should fix your problem.

woyteg, 2023-11-25:
Using Mac fixed the issue. Thank you

closed 2023-11-25 by peterus